wasm-micro-runtime/wasm-micro-runtime #613: Bug with timer using Zephyr OS

AfflatusX, 2021-04-16:
We encountered an issue where a timer would stop working after a period of time while the overall system is still healthy (no crash etc).

After debugging into it, we noticed the sleep param gets really large (3435051 ms) when the bug happens. 

It looks like this function is using k_uptime_get_32() which will overflow after a while: 

https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/core/shared/platform/zephyr/zephyr_time.c#L11

Changing it to k_uptime_get() which provides 64bit version fixed the issue.

xujuntwt95329, 2021-04-20:
Thanks a lot for reporting this issue, we've fixed it in PR #617

wenyongh, 2021-04-20:
@AfflatusX thanks for reporting the issue! It should be caused by integer overflow in k_uptime_get_32() * 1000,  when k_uptime_get_32() is larger than UINT32_MAX/1000, the integer overflow happens. The issue was fixed in #617. Thanks.

AfflatusX, 2021-04-21:
nice! thanks for fixing it!

closed 2021-04-21 by AfflatusX