wasm-micro-runtime/wasm-micro-runtime #615: native stack overflow occurred, native function to recursive call aot export function by "wasm_runtime_call_wasm"

littleblackblack, 2021-04-18:
Running Ubuntu x86_64 in docker.

My aot file is loaded by my iwasm executable program，iwasm invoke native function to recursive call aot export function by wasm_runtime_call_wasm and exception eccurred.

call wasm function exported_aot_inner_func failed. error: Exception: native stack overflow

code snippet:
iwasm impl:
local_call_aot_func_perf_test(wasm_exec_env_t exec_env, int32_t n)
{
    printf("calling local_call_aot_func_perf_test\n");

    uint32_t argv[2] = { 0 };
    wasm_function_inst_t func = NULL;
    if (!(func = wasm_runtime_lookup_function(module_inst, "exported_aot_inner_func",
                                              NULL))) {
        printf(
          "The wasm function exported_aot_inner_func wasm function is not found.\n");
        return;
    }

    uint32_t elpased_ms = 0;
    uint32_t now = wasm_get_sys_tick_ms();
    for (int i = 0; i < n; i++) {
        argv[1] = i;
        if (wasm_runtime_call_wasm(exec_env, func, 2, argv)) {
            // printf(
            //   "Native finished calling wasm function: simple_inner_func, return: %d\n",
            //   result);
        }
        else {
            printf("call wasm function exported_aot_inner_func failed. error: %s\n",
                   wasm_runtime_get_exception(module_inst));
            return;
        }
    }
    elpased_ms = wasm_get_sys_tick_ms() - now;
    printf("eplased %dms\n", elpased_ms);
    printf("finish local_call_aot_func_perf_test\n");
} 

aot file impl:
int32_t
exported_aot_inner_func(int32_t x, int32_t y)
{
    return x + y;
}

When I use wasm instead of AOT, everything works.

littleblackblack, 2021-04-19:
Add that I try to run my program on aarch64 platform and there was no exception occurred. But It takes a long time to run.
I guess this problem is related to this issue https://github.com/bytecodealliance/wasm-micro-runtime/issues/328

wenyongh, 2021-04-20:
Hi @littleblackblack, the AOT mode runs the AOT function like native code, the calling of the AOT functions use the stack of native thread to pass arguments, while in interpreter mode, the WASM function arguments are stored in the WASM operand stack, so normally AOT mode requires more native stack than interpreter mode, and the latter requires more operand stack. It should be normal that "native stack overflow" is thrown in AOT mode when the native stack is exhausted. And it seems that the stack size in your aarch64's platform is bigger so it doesn't throw the exception. Could you please check whether the native stack size is set large enough in x86_64 platform?

From the code of the sample uploaded, there is no recursive call, in fact there is a loop to call the wasm function many times.

closed 2021-06-09 by wenyongh