wasm-micro-runtime/wasm-micro-runtime #621: Can wamr give the ablity to inject exception handle function and do not set excetion any more

xingkaiyu, 2021-04-25:
Hi~
I am using wamr in multi-thread condition. I sometimes need to use detach for long time sub thread and my main thread will finish before the sub thread. Then if the sub thread run into exception, wamr will set excetion itself. But this excetion I will not know and it may cause other calling failed after.
I suggest that wamr can give the chance for us to inject exception handler to the wamr runtime, and pass the exec_env module_inst exception info to this function. Then I can get and handle the exception myself, And you do not need to set the excetion status you self.
Of cause, the result for the call or malloc or other function called from native should be fail (or false).

xujuntwt95329, 2021-04-30:
Hi, I'd like to confirm with you that which kind of handler do you want?

If you just need WAMR to redirect the exception message from `STDOUT` to your handler, then that is easy, we can provide an API to register a global callback for exceptions. Once the exception occurs, the execution will be terminated and the exception message will be passed to your callback.

``` C
wasm_runtime_set_exception_callbak(void (*callback)(exec_env_t exec_env, const char *exception));
```

However, this kind of callback handler can't provide any mechanism to resume the original execution,

xingkaiyu, 2021-06-03:
@xujuntwt95329 
Sorry for relay so late.
I think what I real want can be descriped like this:

1- If wamr can give wasm app the normal exception handle opportunity (normal exception mean the exception wasm app throw it self or other exceptions that can be catch and the wasm app function can goes on):
    In this situation, the handler only need to handle wamr core exception and need these ability:
    a- can give us the exception info: exception name, exception Stack Trace(if has)
    b- can tell the handler whether the exception can be clear and the next call will be ok
    c- if the exception can be clear, I suggest wamr not use wasm_runtime_set_exception any more or we can use 
wasm_runtime_clear_this_exception in the handler to only clear this exception. I do not want this exception cause more fails in other thread or the followed call
   d- if the exception can not be clear, the handler can reload the wasm app by wasm_runtime_load, wasm_runtime_instantiate, wasm_runtime_create_exec_env...

2- If now we can not give the exception handle opportunity to wasm app, then the handler needs to handle all the exception. And this time, we need to know whether the exception is throwed by wasm app it self or by wamr core. Other ability are same with the 1

xujuntwt95329, 2021-06-03:
> 1- If wamr can give wasm app the normal exception handle opportunity (normal exception mean the exception wasm app throw it self or other exceptions that can be catch and the wasm app function can goes on):
In this situation, the handler only need to handle wamr core exception and need these ability:
a- can give us the exception info: exception name, exception Stack Trace(if has)
b- can tell the handler whether the exception can be clear and the next call will be ok
c- if the exception can be clear, I suggest wamr not use wasm_runtime_set_exception any more or we can use
wasm_runtime_clear_this_exception in the handler to only clear this exception. I do not want this exception cause more fails in other thread or the followed call
d- if the exception can not be clear, the handler can reload the wasm app by wasm_runtime_load, wasm_runtime_instantiate, wasm_runtime_create_exec_env...

Seems you are considering [exception handling proposal](https://github.com/WebAssembly/exception-handling), we are also evaluating this proposal. However, as far as I know, most of the compilation toolchains are not ready for this feature. For example `AssemblyScript` will simply emit an `abort` when throwing exception, and `wasi-sdk` will report a compilation error if there are exceptions in the C++ code.

> 2- If now we can not give the exception handle opportunity to wasm app, then the handler needs to handle all the exception. And this time, we need to know whether the exception is throwed by wasm app it self or by WAMR core. Other ability are same with the 1

Currently all the exceptions are from runtime, wasm app don't have the ability to throw exceptions without `exception handling` proposal

xingkaiyu, 2021-06-08:
e~~~~
So the only why is wait....