wasm-micro-runtime/wasm-micro-runtime #625: I can't compile WAMR in MIPS+MUSL(openwrt) environment

cooboyno111, 2021-05-07:
I tried to write a makefile to bring WAMR into openwrt。
This makefile works well in the environment of aarch64 + MIPS + musl。
However, mipsel (mt7621) can not work normally。

/home/lup/Desktop/kunteng-lede-17.01.4/build_dir/target-mipsel_24kc_musl-1.1.16/wamr/core/iwasm/common/arch/invokeNative_mips.s:58: Error: opcode not supported on this processor: mips1 (mips1) `ldc1 $f12,0($sp)'
/home/lup/Desktop/kunteng-lede-17.01.4/build_dir/target-mipsel_24kc_musl-1.1.16/wamr/core/iwasm/common/arch/invokeNative_mips.s:59: Error: opcode not supported on this processor: mips1 (mips1) `ldc1 $f14,8($sp)'

I searched for the question that might be related to __mips_soft_float Please give your official support。

wenyongh, 2021-05-08:
Hi, it seems that float point register isn't supported on the processor and it is unnecessary to copy float point registers from stack, could you please remove line 58, 59 and try again?

cooboyno111, 2021-05-10:
My platform can't use this solution。（invokeNative_mips.s）
But this platform can use this solution。（invokeNative_general.c）
However, cmakelist.txt in the root directory does not support this platform.（GENERAL）
I direct modify core/iwasm/common/iwasm_common.cmake。
line 52 to   set (source_all ${c_source_all} ${IWASM_COMMON_DIR}/arch/invokeNative_general.c)
Please add the support of general platform directly to cmakelist.txt in the root directory.

wenyongh, 2021-05-12:
Hi, building the invokeNative_general.c explicitly is supported now, please pull the latest code and use "cmake -DWAMR_BUILD_INVOKE_NATIVE_GENERAL=1 ..." to build it.

closed 2021-05-27 by wenyongh

ThomasDevoogdt, 2024-01-22:
@wenyongh this issue was also seen while adding support on buildroot for fluent-bit, please refer to this commit: https://github.com/buildroot/buildroot/commit/9b19030f3f7505af8c026cb63f0f4f509836d1e9. Would it make sense to set this flag by default for MIPS?

wenyongh, 2024-01-23:
Hi, I saw the issue has been resolved by adding build option `FLUENT_BIT_CONF_OPTS += -DWAMR_BUILD_INVOKE_NATIVE_GENERAL=1`, is it good enough? Had better not set the flag by default MIPS by default since it has limitation:
https://github.com/bytecodealliance/wasm-micro-runtime/blob/b44aa654b8f41112dd02a84c2e3246e669e818b6/core/iwasm/common/iwasm_common.cmake#L31-L36

ThomasDevoogdt, 2024-01-23:
> Hi, I saw the issue has been resolved by adding build option `FLUENT_BIT_CONF_OPTS += -DWAMR_BUILD_INVOKE_NATIVE_GENERAL=1`, is it good enough? Had better not set the flag by default MIPS by default since it has limitation:

Ok, fine enough, main problem was an unknown opcode 'ldc1', seen on mips32r2 and mips64r2, so at max, CMake can verify that upfront, bot not a priority for me.