wasm-micro-runtime/wasm-micro-runtime #626: Question about wasm sub thread

bxkai, 2021-05-10:
When calling pthread_create() to create a thread, sub thread call native api xxx as below. The module instance from exec env is different with the main thread‘s module instance.

int xxx_native(wasm_exec_env_t exec_env , ...)
{
    ...
}

The function wasm_exec_env_get_module can get same wasm module, but this function is not exported. Can you add this function to wasm_export.h? Thanks！

wustwn, 2021-05-10:
Hey, actually WAMR has exported the function to get the module instance, which is `wasm_runtime_get_module_inst`.  Could you please have a try?

bxkai, 2021-05-10:
@wustwn sub thread get the module instance is oneself, but not the main thread‘s module instance.
When calling pthread_create() to create a thread, wamr([pthread_create_wrapper](https://github.com/bytecodealliance/wasm-micro-runtime/blob/9710d9325f426121cc1f2c62386a71d0c022d613/core/iwasm/libraries/lib-pthread/lib_pthread_wrapper.c#L545)) create a new module instance and new exec env!

wustwn, 2021-05-10:
`wasm_runtime_get_module_inst`  just invokes the ` wasm_exec_env_get_module_inst`

```c
WASMModuleInstanceCommon * wasm_runtime_get_module_inst(WASMExecEnv *exec_env)
{
    return wasm_exec_env_get_module_inst(exec_env);
}
```

so it should be the same as the `wasm_exec_env_get_module_inst`. Not sure whether you want to get the sub thread's module instance or main thread's module instance.

bxkai, 2021-05-12:
@wustwn  we save some user data according to exec env（main thread’s);  but the sub thread create int wasm, the exec env is different with the main thread, we can not get user data

xujuntwt95329, 2021-05-19:
Hi @bxkai,

We just support spreading custom data through all threads in PR #633 , could you please use `wasm_runtime_set_custom_data` to attach your user data to the `module_instance`, and then you can use `wasm_runtime_get_custom_data` in any sub-threads from the instance

closed 2021-05-21 by bxkai