wasm-micro-runtime/wasm-micro-runtime #632: call wasm_runtime_module_malloc Exception: out of memory

victortanjh, 2021-05-15:
I call a wasm function from native world, and pass parameter which is a buffer.
this is my code for malloc buffer:
    buffer_for_wasm = wasm_runtime_module_malloc(module_inst, txidLen,(void**)&bufferInput);
    if (buffer_for_wasm == 0 ){
        LogErrorf("Malloc failed in wasm:%s\n",wasm_runtime_get_exception(module_inst));
        return;
    }

I often got a error log: Malloc failed in wasm:Exception: out of memory.

I don't know the reason, buf , I think it is releated to the wasm code.
My wasm source code language is AssemblyScript, and there's a code snippet:
let keyBuf = new ArrayBuffer(VALUE_BUF_LEN);

If the VALUE_BUF_LEN==1024*10, the error “out of memory”  occurs frequently， 
but, if VALUE_BUF_LEN==1024，the error no longer occurs.

the heap size and stack size is :
#define WASM_GLOBAL_HEAP_SIZE (1024 * 1024 * 128)
#define WASM_MODULE_INSTANCE_HEAP_SIZE (1024*1024)
#define WASM_MODULE_INSTANCE_STACK_SIZE (1024*1024)

Does anyone know the reason?

Thanks a lot.

wenyongh, 2021-05-17:
Hi, could you please add `-Wl,--export=malloc -Wl,--export=free` option for wasi-sdk so that module_malloc can allocate the memory from the libc heap instead of app heap initialized by runtime?

victortanjh, 2021-05-18:
I just use the "iwasm VM core" as libvmlib.a to be integrated to my application.
How to add " -Wl,--export=malloc -Wl,--export=free " to the compilation process of "iwasm VM core" ? 
Add it to the CMakeLists.txt？
thank you.

 my AssemblyScript code is as follows:
export  function func1(){
   const VALUE_BUF_LEN :i32 = **1024*10;**
    for (i=0; i< 100; i++){
        value = func1(txid, key);
    }
}

function func1(txid:string, key:string):ArrayBuffer{
   let valueBuf = new ArrayBuffer(VALUE_BUF_LEN);
   return valueBuf;
}

After I call the above wasm code once, then exec wasm_runtime_module_malloc  in the native world will be fail.
But, if I change the const VALUE_BUF_LEN :i32 = **1024*5;**
then in the native world, calling wasm_runtime_module_malloc will success.

And, setting this compilation macro does not work:
#define WASM_GLOBAL_HEAP_SIZE (1024 * 1024 * 128)
#define WASM_MODULE_INSTANCE_HEAP_SIZE (1024 * 1024)
#define WASM_MODULE_INSTANCE_STACK_SIZE (1024 * 1024)

wenyongh, 2021-05-19:
Hi, for AssemblyScript, which version of asc do you use? And are these functions exported in your wasm files: __new/__retain/__release (or __new/__pin/__unpin for older asc version)? You can use wabt tool to check it, and if not, could you please export them with "--exportTable" option and try again?

victortanjh, 2021-05-21:
I just use "npm run asbuild" to compile the AssemblyScript code.
the config is as follows:
**package.json:**
  "dependencies": {
    "@assemblyscript/loader": "^0.18.32",
    "assemblyscript-json": "^1.0.0"
  },
  "devDependencies": {
    "assemblyscript": "^0.18.32"
  }

**asconfig.json:**
{
  "targets": {
    "debug": {
      "binaryFile": "build/untouched.wasm",
      "textFile": "build/untouched.wat",
      "sourceMap": true,
      "debug": true
    },
    "release": {
      "binaryFile": "build/optimized.wasm",
      "textFile": "build/optimized.wat",
      "sourceMap": true,
      "optimizeLevel": 3,
      "shrinkLevel": 1,
      "converge": false,
      "noAssert": false
    }
  },
  "options": {
  **"exportTable": true**
  }
}

Is this the right way to add "--exportTable" option?

wenyongh, 2021-05-24:
Hi, I have a test, the "--exportTable" is to export the function table with exported symbol named "table", the correct option should "--exportRuntime":
```bash
--exportRuntime       Exports the runtime helpers (__new, __collect etc.).
```
So please add `"exportRuntime": true` instead of  `"exportTable": true`. Thanks.

closed 2021-08-20 by wenyongh