wasm-micro-runtime/wasm-micro-runtime #634: Is it possible to build wamrc which running on linux from mac?

JavanZhu, 2021-05-19:
Hi, My develop environment is mac, and I need to build wamrc which running on linux or android, so is it possible to build wamrc which running on linux from mac? Thanks.

xujuntwt95329, 2021-05-19:
Of course you can, this is something just like cross-compilation, you can use `--target` option to specify a triple. 

However, I'd suggest you using docker, it would be more convenient

JavanZhu, 2021-05-20:
@xujuntwt95329 Thank you for quick support, I have built wamrc successfully for linux platform on mac using docker. 
And I am also trying to build wamrc for android, I understand that maybe we should rebuild llvm and wamrc for cross-compilation depends on 
>  you can use --target option to specify a triple.

 but I didn't find any way or document about how to build wamrc for android platform, could you please give me any suggestion about it. thank you.

xujuntwt95329, 2021-05-20:
You should use the toolchain from `Android SDK` to build both LLVM and wamrc, you may need to write some `.mk` file to control the source, include and linking path. I'm sorry that I can't provide the complete steps as I'm not very familiar with Android toolchain.

JavanZhu, 2021-05-20:
@xujuntwt95329 Ok, Thank you very much for your help. I'll try it

closed 2021-05-20 by JavanZhu