wasm-micro-runtime/wasm-micro-runtime #642: `wasm_frame_t` Methods are defined but not implemented

AlexEne, 2021-05-26:
`wasm_c_api.h` exports the following methods:

```
WASM_DECLARE_OWN(frame)
WASM_DECLARE_VEC(frame, *)
WASM_API_EXTERN own wasm_frame_t* wasm_frame_copy(const wasm_frame_t*);

WASM_API_EXTERN struct wasm_instance_t* wasm_frame_instance(const wasm_frame_t*);
WASM_API_EXTERN uint32_t wasm_frame_func_index(const wasm_frame_t*);
WASM_API_EXTERN size_t wasm_frame_func_offset(const wasm_frame_t*);
WASM_API_EXTERN size_t wasm_frame_module_offset(const wasm_frame_t*);
```

I was planning in using them, along with: `wasm_trap_origin(const wasm_trap_t*);` to dump a stack trace from the VM.

Trying to use them in a program will result in a compilation error that the functions aren't defined.

lum1n0us, 2021-05-27:
YES. WAMR doesn't implement DEBUG related feature in `wasm_c_api.h` at the moment.

AlexEne, 2021-05-27:
I see that dumps to `stdout` would be interesting to make it call a callback instead of `printf`. 
Registering a callback for this purpose may help, as people may want to capture it and report it somewhere other than with a printf.

lum1n0us, 2021-05-27:
🤔 maybe we can do something

AlexEne, 2021-05-27:
Just for clarity, are you planning in doing some work for this or is it fine for me to try and follow up with a PR that implements something along these lines?

wenyongh, 2021-05-29:
@AlexEne WAMR supports redirect the stdout output to the callback function provided by you, please refer to:

https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/doc/build_wamr.md#wamr_bh_vprintfvprintf_callback-default-to-disable-if-not-set

AlexEne, 2021-05-29:
I think it may be good a work-around. One would need to filter all the calls somehow to find the callstack for a crash and store it / report it separately but it seems like a good solution until the frame functions are exposed.

t-veor, 2021-06-29:
@wenyongh Hello, I see that this ticket is set to work-in-progress - is there someone currently working on this issue? I can help by taking a look at the implementation if someone isn't already working on this.

wenyongh, 2021-07-05:
@t-veor Thanks a lot! We are implementing these APIs and have basically enabled them for interpreter mode, the patch will be submitted soon and we will continue to enable the AOT mode then.

t-veor, 2021-07-13:
Hi @wenyongh, I see you've raised and merged patches #660 and #663 for this issue. I was experimenting with the APIs but I see in interpreter mode the frame data comes from here in wasm_runtime.c:
```
        /* place holder, will overwrite it in wasm_c_api */
        frame.instance = module_inst;
        frame.module_offset = 0;
        frame.func_index = func_inst - module_inst->functions;
        frame.func_offset =
          cur_frame->ip ? cur_frame->ip - func_inst->u.func->code : 0;
```
For the use case of dumping a stack trace it would be useful to know the byte offsets of the instruction that caused the trap and all the function call instructions in the other frames. Is it possible to recover this byte offset somehow? It looks like there are three obstacles with using this information:

* It looks like `cur_frame->ip - func_inst->u.func->code` is wrong, because in fast interpreter mode `cur_frame->ip` actually points into the compiled code array and not the code array.
* In fast interpreter mode the IP doesn't correspond to byte offsets anyway unless there is a mapping back to the original bytecode from the compiled code
* There seems to be no way to recover a list of functions in order to calculate the location of a function from its `func_offset`

Does such a mapping exist so the offset information for the stack trace can be recovered?

In AOT mode I see the frame data comes from here in aot_runtime.c:
```
        frame.instance = module_inst;
        frame.module_offset = 0;
        frame.func_index = cur_frame->func_index;
        frame.func_offset = 0;
```
which would mean that I won't be able to recover anything except the function the trap occurs in and all the calling functions. Is there no way to map this back to a location in the bytecode? I can take a look and help with the implementation.

lum1n0us, 2021-07-14:
you are right about the `func_offset` in *fast-interpreter* mode.  It should be `code_compiled`. 

about AOT mode, we'd like to hear your idea and be glad to have your help.

lum1n0us, 2021-07-29:
The [latest commit](https://github.com/bytecodealliance/wasm-micro-runtime/commit/edb184f709b83fdae46835be792ca9a31dc5ae41) includes more missing *wasm-c-api* functions. You might want to take a look. Good day.

wenyongh, 2022-02-24:
Close the issue as the missing APIs had been implemented.

AlexEne, 2022-02-24:
Thanks for the help!

closed 2022-02-24 by AlexEne