wasm-micro-runtime/wasm-micro-runtime #646: Missing wasm-c-api functions

clover2123, 2021-06-02:
Hi all!

I'm trying to import wamr runtime using `wasm-c-api`.
But I found that there are some missing api as follows.

* wasm_ref_t functions
* bool wasm_module_validate(wasm_store_t*, const wasm_byte_vec_t* binary)
* wasm_memory_pages_t wasm_memory_size(const wasm_memory_t*)
* bool wasm_memory_grow(wasm_memory_t*, wasm_memory_pages_t delta)
* wasm_ref_t* wasm_table_get(const wasm_table_t*, wasm_table_size_t index)
* bool wasm_table_set(wasm_table_t*, wasm_table_size_t index, wasm_ref_t*)
* wasm_table_size_t wasm_table_size(const wasm_table_t*)
* bool wasm_table_grow(wasm_table_t*, wasm_table_size_t delta, wasm_ref_t* init)

It would be great if you support the above features.
Thanks a lot!

lum1n0us, 2021-06-03:
thanks for asking. we'll make a plan about it.

BTW,  would you mind sharing your project information with us?  because those APIs are rarely used by others and I am curious about the scenario.

clover2123, 2021-06-03:
> BTW, would you mind sharing your project information with us? because those APIs are rarely used by others and I am curious about the scenario.

Sure, I'm updating [wasm-js interface](https://webassembly.github.io/spec/js-api/) spec in our JavaScript engine.
(wabt is currently used but it's performance is too low. So, I searched other wasm engines and found that wamr is much more faster than wabt.)

According to the wasm-js specification, JavaScript built-in Objects such as `Module, Instance, Memory, Table and Global` represent wasm object each and should support its relative operations.
For example, `Table` object has a wasm table address internally and could set/get table members.

andrewrch, 2021-06-22:
Jumping on this ticket rather than making a new one - there is also ``wasm_engine_new_with_args`` which I can't find in a header, and is crucial if we want to use our own allocator.

I would have opened a PR for this, but I saw this issue is marked WIP so didn't want to conflict (Also the parameters are defined in ``wasm_export.h`` and I wasn't sure whether to move them to another header).

wenyongh, 2021-07-15:
@andrewrch wasm_engine_new_with_args() and the related structures are declared in wasm_c_api.h, please include it if you want to call this function, thanks.

lum1n0us, 2021-07-29:
The [latest commit](https://github.com/bytecodealliance/wasm-micro-runtime/commit/edb184f709b83fdae46835be792ca9a31dc5ae41) includes more missing *wasm-c-api* functions. You might want to take a look. Good day.

clover2123, 2021-07-29:
@lum1n0us 
Thanks a lot! It seems that most of the `wasm-c-api` functions are now supported.
BTW, according to the [wasm-c-api doc](https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/doc/wasm_c_api.md)
> we tend to grow a table or a memory by opcode only and not support growing both by host-side function callings.

You don't have a plan for grow api such as `wasm_table_grow, wasm_memory_grow`, right?

lum1n0us, 2021-07-29:
We will treat it as a design limitation until implementing the GC feature.