wasm-micro-runtime/wasm-micro-runtime #650: How to get the return value of the WASM main function?

ya0guang, 2021-06-08:
Hi,

I just encountered a problem while embedding WAMR. I want to get the return value from the `main` function of the web assembly binary but failed with `wasm_application_execute_main`. Unlike `wasm_runtime_call_wasm` which puts the return value of the function in WASM in `wasm_argv[0]`, it seems that  `wasm_application_execute_main` doesn't put any return value back to the input arguments(`argv[0]`).

My snippet of calling the function:
```c
    /*Call the main function of the WASM*/
    int wasm_argc = 2;
    int* wasm_argv[2] = {"2", "3"};
    printf("before exec: %d, %d, %s, %s\n", wasm_argv[0], wasm_argv[1], wasm_argv[0], wasm_argv[1]);
    int ret = (int)wasm_application_execute_main(module_instance, wasm_argc, wasm_argv);
    printf("after exec: %d, %d, %s, %s\n", wasm_argv[0], wasm_argv[1], wasm_argv[0], wasm_argv[1]);
```

The very simple main function in C compiled to WASM:
```c
int main(int argc, char *argv[])
{   
    return 19;
    // return atoi(argv[1]);
}
```

And the output:
> before exec: 283970230, 283967786, 2, 3
> after exec: 283970230, 283967786, 2, 3

It seems nothing is returned by `wasm_application_execute_main`. Is there a way to get the return value out from the WASM main function?

Thanks!

wenyongh, 2021-06-16:
@ya0guang thanks for reporting the issue! It was fixed in patch #652, please get the return value of main function by `*(int*)argv`.

ya0guang, 2021-07-20:
Sorry for the late reply.

Thanks a lot for fixing this!

closed 2021-07-20 by ya0guang