wasm-micro-runtime/wasm-micro-runtime #666: Clang -msimd128 compiled wasm generates error when run

robinvanemden, 2021-07-19:
When trying to run a `clang` compiled (with `-msimd128` command line argument) wasm with a `wamr` binary (using `WAMR_BUILD_SIMD 1`) running in Unbunu X86_64/Skylake I receive:

`WASM module load failed: v128 value type requires simd feature`

Is it not (yet?) possible to run "pure" clang generated wasm binaries with `msimd128`, or am I missing an essential ingredient?

wenyongh, 2021-07-19:
Hi, SIMD is only supported in AOT mode and JIT mode, please build iwasm with JIT mode to run the wasm file, or [build wamrc](https://github.com/bytecodealliance/wasm-micro-runtime#build-wamrc-aot-compiler) to comple was file into aot file, and run the aot file, e.g.:
```bash
wamrc -o test.aot test.wasm
iwasm test.aot
```

robinvanemden, 2021-07-19:
> ```shell
> wamrc -o test.aot test.wasm
> ```

In the latter case I get:
```
./wamrc -o mnist-clang-simd.wasm mnist-clang-simd.wasm
WASM module load failed: type mismatch: expect v128 but got other
```

robinvanemden, 2021-07-19:
> I did use a precompiled version of LLVM 10.0. I will try again with the LLVM 11.0 based compile script, and let you know if that resolved the issue.

I updated both WAMR and wasi-sdk LLVM's to version 11, which resolved the issue. Thanks for your fast response!

closed 2021-07-19 by robinvanemden