wasm-micro-runtime/wasm-micro-runtime #668: Can't find file 'os_api.h'

hsyhhh, 2021-07-21:
Hi!

I was trying to run the sample application `wamr_esp_idf` (which is provided in directory `/product-mini/platforms/esp-idf` ) on board `esp32-wroom-32`, and I built the project in Ubuntu 18.04. However, when I tried to run cmake in this directory, it reported the following error:

```
In file included from /home/hsy/wamr/wasm-micro-runtime/core/shared/platform/esp-idf/../include/platform_common.h:13,
                 from /home/hsy/wamr/wasm-micro-runtime/core/shared/platform/esp-idf/../include/platform_api_vmcore.h:9,
                 from /home/hsy/wamr/wasm-micro-runtime/core/shared/platform/esp-idf/espidf_thread.c:6:
/home/hsy/wamr/wasm-micro-runtime/core/shared/platform/esp-idf/platform_internal.h:21:10: fatal error: os_api.h: No such file or directory
 #include "os_api.h"
          ^~~~~~~~~~
compilation terminated.
```

It seems that in `/core/shared/platform/esp-idf/platform_internal.h`, it includes some files in `esp-idf` as follows: (I changed `#include <FreeRTOS.h>` to `#include "freertos/FreeRTOS.h"` so that cmake can find the first three files)

```c
// in file: /core/shared/platform/esp-idf/platform_internal.h, line 18
#include "freertos/FreeRTOS.h"
#include "freertos/semphr.h"
#include "freertos/task.h"
#include "os_api.h"
```

I find the first three files in `esp-idf`, whose directory is `esp-idf/components/freertos/include/freertos`. But I can't find `os_api.h` in `esp-idf` or in `wasm-micro-runtime`.

My ESP-IDF version is v4.0. The shell script used to build the application `wamr_esp_idf` are as follows:

```shell
# in dir: ~/wamr/wasm-micro-runtime/product-mini/platforms/esp-idf
rm -rf build && mkdir build && cd build
cmake .. -DCMAKE_TOOLCHAIN_FILE=$IDF_PATH/tools/cmake/toolchain-esp32.cmake -DTARGET=esp32 -GNinja
cmake --build .
```

I don't where it went wrong, which bothers me a lot. Thanks a lot!

closed 2021-08-07 by hsyhhh

1c3t3a, 2021-12-10:
Hi there! I am running into the same problem. Did you find a solution?

hsyhhh, 2021-12-10:
> Hi there! I am running into the same problem. Did you find a solution?

Sorry, still not solved.