wasm-micro-runtime/wasm-micro-runtime #671: Build failed by wamr_toolchain.cmake

luyahan, 2021-07-23:
I follow this to build hello_world:
```
We can generate a CMakeLists.txt file for test.c:

cmake_minimum_required (VERSION 3.5)
project(hello_world)

set (CMAKE_EXE_LINKER_FLAGS "${CMAKE_EXE_LINKER_FLAGS},--export=main")
add_executable(hello_world test.c)
It is simple to build this project by cmake:

mkdir build && cd build
cmake .. -DCMAKE_TOOLCHAIN_FILE=$WAMR_ROOT/wamr-sdk/app/wamr_toolchain.cmake
make
```
encounter errors:
```
luyahan@p9-plct:~/source/wasm-micro-runtime/wamr-sdk/test $ make
[ 50%] Linking C executable hello_world
wasm-ld: error: symbol exported via --export not found: main
clang-11: error: linker command failed with exit code 1 (use -v to see invocation)
CMakeFiles/hello_world.dir/build.make:94: recipe for target 'hello_world' failed
make[2]: *** [hello_world] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/hello_world.dir/all' failed
make[1]: *** [CMakeFiles/hello_world.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2
```

OS:Ubuntu
wasi-sdk: wasi-sdk-12.0-linux

xujuntwt95329, 2021-07-23:
Hi, the wasi-sdk version 12 treat the "main" function as `__main_argc_argv`, could you please change this line 
``` cmake
set (CMAKE_EXE_LINKER_FLAGS "${CMAKE_EXE_LINKER_FLAGS},--export=main")
```
in CMakeLists.txt to
``` cmake
set (CMAKE_EXE_LINKER_FLAGS "${CMAKE_EXE_LINKER_FLAGS},--export=__main_argc_argv")
```
and try again?

luyahan, 2021-07-23:
> Hi, the wasi-sdk version 12 treat the "main" function as `__main_argc_argv`, could you please change this line
>
> ```cmake
> set (CMAKE_EXE_LINKER_FLAGS "${CMAKE_EXE_LINKER_FLAGS},--export=main")
> ``` [...]

Thanks 
I build pass

closed 2021-07-23 by luyahan