wasm-micro-runtime/wasm-micro-runtime #679: How to limit cpu consumption of wasm instance?

victortanjh, 2021-07-31:
for example, my wasm bytecode compiled from AssemblyScript。
For a test, there is a while loop in the AssemblyScript code:
function infiniteLoop(): Response {
    while(1){
        smlog("This is infinite loop,getAccount finished, the value is :"+ArrayBuffer2Str(value));
    }
}

The whole process is blocked and can't recover. 

My question is : Can we limit the cpu consumption  of wasm instance ? Or, can we kill the wasm instance after a while.

Thank you!

xwang98, 2021-07-31:
We can do that by adding flag checking in the branch for loop in the runtime, and use additional thread for setting the flag when the guard time expires, the runtime raises exception for the Wasm app when the flag is set. However it will introduce a big performance penalty.

If we only want to check for specific loops, we can use a new opcode like “guarded_branch” to replace the branch opcode for the loop as we have implemented this way in the fast interpreter. The new opcode will be checked for the flag. And we need to find a way to let developer annotate those loops for guard.

victortanjh, 2021-07-31:
sorry, I still do not know how to do. 
Now, in my app, the libvmlib.a is linked, and I use wasm_runtime_call_wasm function to call the function in AssemblyScript.
Can I call a API of libvmlib.a to set the timeout of one wasm function call?

xwang98, 2021-07-31:
@victortanjh This function is not supported in WAMR yet. My last post is just one idea to implement your requirement. Let's put this requirement into the want list.

victortanjh, 2021-08-02:
ok. Thank you very much.

stgatilov, 2021-11-13:
> We can do that by adding flag checking in the branch for loop in the runtime, and use additional thread for setting the flag when the guard time expires, the runtime raises exception for the Wasm app when the flag is set. However it will introduce a big performance penalty.
Would it be a performance problem because of multithreading?

What about adding some kind of callback which is called on regular basis? Would it be a performance problem even if the callback is empty?
The user can write whatever he wants in the callback: checking atomic boolean flag, or merely comparing current time against limit once per 1024 calls. That's how Lua does it (see lua_sethook and LUA_MASKCOUNT).
I think there are users who absolutely need to protect against hanging/blocking, and they can accept the performance penalty. It's more the problem for those who don't need protection.

**P.S.** For the context, I organize an annual contest for student programmers, and the funniest thing in this contest is a massive programming game. You can see [one example here](http://codeforces.com/blog/entry/48236). Each participant implements a program which control his player, and all these programs run simultaneously in real time. Basically, we need a way to run 50-100 simple sandboxed programs and run their "think" function 60 times per second on single core of an ordinary PC. Lua does this perfectly, except that we have to force all students to learn it. The holy grail would be to allow programs in C++, and that's why I'm here =)