wasm-micro-runtime/wasm-micro-runtime #680: Is `wasm_runtime_module_malloc` thread-safe?

vdimir, 2021-07-31:
I get strange falires when calling `wasm_runtime_module_malloc` with different module instances from multiple threads, does it mean that this function is not thread safe? I use `Alloc_With_Allocator` option in `RuntimeInitArgs`. It's expected that in may be not thread safe with `Alloc_With_Pool`, but if if calls `malloc` it should be safe or not?

xujuntwt95329, 2021-08-01:
Hi, how do you build your wasm application ?

If your wasm application are built as `LIBC-WASI` mode, you need to hack the `wasi-libc`'s dlmalloc.o as the `dlmalloc` function implemented by wasi-libc is not thread-safe, please refer to `Build with libc-WASI` section in [pthread_library.md](https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/doc/pthread_library.md#wamr-pthread-library) for more details.

If you are using `LIBC-BUILTIN` mode, then `wasm_runtime_module_malloc` is thread-safe.

vdimir, 2021-08-02:
> Hi, how do you build your wasm application ?

I compile it with Emscripten `emcc testapp.c -s STANDALONE_WASM -s ERROR_ON_UNDEFINED_SYMBOLS=0 -O2 -s TOTAL_MEMORY=6MB -o testapp.wasm --no-entry`, but actually no explicit allocations are performed in wasm app.

Ok, I'll try different emcc options or another compiler.

closed 2022-09-27 by wenyongh