wasm-micro-runtime/wasm-micro-runtime #682: the WAMR_BUILD_MEMORY_PROFILING option has not take effect

victortanjh, 2021-08-02:
use the following command to compile WAMR vmcore (iwasm) :
cmake .. -WAMR_BUILD_DUMP_CALL_STACK=1 -WAMR_BUILD_MEMORY_PROFILING=1
make

then, I use the following command to find the "wasm_runtime_dump_mem_consumption" function, but failed.

nm -g --defined-only libvmlib.a |grep wasm_runtime_dump_mem_consumption

And also, when I use the wasm_runtime_dump_mem_consumption in my code, I got the following error:

undefined reference to `wasm_runtime_dump_mem_consumption'

wenyongh, 2021-08-02:
@victortanjh, please use -**D**WAMR_BUILD_DUMP_CALL_STACK=1 -**D**WAMR_BUILD_MEMORY_PROFILING=1, but not -WAMR_BUILD_DUMP_CALL_STACK=1 -WAMR_BUILD_MEMORY_PROFILING=1, thanks.

victortanjh, 2021-08-02:
thank you very much! @wenyongh

closed 2022-02-17 by wenyongh