wasm-micro-runtime/wasm-micro-runtime #684: the call of wasm function can be aborted by destroy wasm instance?

victortanjh, 2021-08-03:
In the AssemblyScript function, there is a infiniteLoop function.
function infiniteLoop(): Response {
    while(1){
        smlog("This is infinite loop,getAccount finished, the value is :"+ArrayBuffer2Str(value));
    }
}
Then, when I call the  infiniteLoop function from native world,  it will never return.
I call the following function to stop the call of infiniteLoop,but my app crash:
wasm_runtime_destroy_exec_env(wasmCtx->exec_env);
wasm_runtime_deinstantiate(wasmCtx->module_inst);
wasm_runtime_unload(wasmCtx->wasmModule);

My question is, can I stop the call of wasm function(like infiniteLoop), and not crash?

wenyongh, 2021-08-03:
@victortanjh WAMR provides multi-thread support, the related APIs are defined in core/iwasm/libraries/thread_mgr.h, if you want to terminate the thread that wasm app is running from another thread (e.g. native thread), could you please try calling `int32 wasm_cluster_cancel_thread(WASMExecEnv *exec_env)`? Note that the multi-thread feature should be enabled: use `cmake -DWAMR_BUILD_LIB_PTHREAD=1 ....` to build iwasm, and if you want to run in AOT mode, pelase use `wamrc --enable-multi-thread ....` to generate the AOT file.

victortanjh, 2021-08-04:
Thank you very much!
@wenyongh

closed 2021-12-30 by wenyongh