wasm-micro-runtime/wasm-micro-runtime #685: Exception: failed to call unlinked import function (env, abort)

victortanjh, 2021-08-04:
When I calling a wasm function(AssemblyScript) by wasm_runtime_call_wasm(exec_env, func, 2, argvIn), I got a error.
I catch the exception by wasm_runtime_get_exception(module_inst), and the exception info is "Exception: failed to call unlinked import function (env, abort)".

My question is : can I get the call stack in AssemblyScript?  And what's the meaning of "Exception: failed to call unlinked import function (env, abort)"?

Thank you very much!

wustwn, 2021-08-04:
@victortanjh please ref to #510 , runtime cannot find the `abort function` in `libc-builtin` by default, so you should implement your `abort wrapper` native API and [register it to runtime](https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/doc/export_native_api.md).  

Could you please have a try? Thanks.

victortanjh, 2021-08-04:
Thank you! I followed the #510 .
This is my abort wrapper function:
void AbortWrapper(wasm_exec_env_t exec_env, char *msg, char *fileName, uint32_t lineNumber, uint32_t columnNumber)
{
     LogInfof("enter AbortWrapper");
     wasm_module_inst_t module_inst = get_module_inst(exec_env);
     char buf[2048];
     snprintf(buf, sizeof(buf), "env.abort(msg:%s, file:%s, lineNumber:%i, columnNumber:%i)",msg, fileName, lineNumber, columnNumber);
     wasm_runtime_set_exception(module_inst, buf);
}

this is the native_sysbols register info:
{"abort", AbortWrapper, "($$ii)", NULL},

and the catched exception is :
Exception: env.abort(msg:, file:~, lineNumber:268, columnNumber:14)
**but, the msg and file info is empty, do you known why?**

wustwn, 2021-08-04:
@victortanjh  It seems that the `msg` here is probably a struct rather than a string, so maybe you should check the details of it. If the `msg` is a struct, you should use `*` to replace the `$` when registering API to runtime.

Can you have a try? Thanks.

victortanjh, 2021-08-05:
Thank you!

the definition of abort in AssemblyScript is:
// @ts-ignore: decorator
@external("env", "abort")
declare function abort(
  message?: string | null,
  fileName?: string | null,
  lineNumber?: u32,
  columnNumber?: u32
): void;

So, maybe this is a bug for AssemblyScript.

xujuntwt95329, 2021-08-11:
Hi @victortanjh 

This may not be a bug, I've checked the [document](https://www.assemblyscript.org/stdlib/string.html#string) from AssemblyScript, the String in AS is encoded as `UTF-16`, which is not compatible with C string. You may need to decode it into `ASCII` in the wrapper, or you can override the default abort function provided by AS and pass a `UTF-8` encoded string to your wrapper function, please refer to [override abort](https://www.assemblyscript.org/debugging.html#overriding-abort)

Note: you can convert AS string to `UTF-8` encoded one by this [encoding API](https://www.assemblyscript.org/stdlib/string.html#encoding-api):
``` javascript
function String.UTF8.encode(str: string, nullTerminated?: bool): ArrayBuffer
```