wasm-micro-runtime/wasm-micro-runtime #686: Errors in running samples on AArch64 Ubuntu18.04

xiangtianheng, 2021-08-04:
Trying to run some samples given in the repo, https://github.com/bytecodealliance/wasm-micro-runtime/tree/main/samples/basic and https://github.com/bytecodealliance/wasm-micro-runtime/tree/main/samples/multi-module

Both seem to have the same error. 

In the "basic" sample, when I ./build.sh, this error occurs: "error: unrecognized command line option '-mindirect-branch-register'", then I comment line 44 in the CMakeLists.txt of the "basic" sample, "error: 'MAP_32BIT' undeclared (first use in this function); did you mean 'MMAP_MAP_32BIT'?" is prompted. The same will happen when I am trying to compile the "multi-module" sample.

wustwn, 2021-08-05:
@xiangtianheng Could you please change the `set (WAMR_BUILD_TARGET "X86_64")`  to `set (WAMR_BUILD_TARGET "AARCH64")` in the CMakeLists.txt and try again. Thanks.

xiangtianheng, 2021-08-05:
Thank you! It turns out that I miss this setting in the CMakeLists.

closed 2021-08-05 by xiangtianheng