wasm-micro-runtime/wasm-micro-runtime #687: Is there a "iwasm" for ARM？

yylittlefish, 2021-08-05:
Hello~
For linux platform, the source code provide a method to build mini product "iwasm" and launch .wasm file.
for arm architecture, I found there was a sample "simple", 
but it use a remote control to install/uninstall wasm app.
is there a "iwasm" type binary file I could generate and use it in arm in this way: iwam test.wasm?

Thank you!

yylittlefish, 2021-08-05:
I found the method, build iwasm in /product-mini/platforms/linux/folder with toolchain.cmake will be okay!

wustwn, 2021-08-05:
@yylittlefish 
Please make sure that cross compiler `arm-linux-gnueabi` has been installed correctly on your platform.

Then add following configs into the `product-mini/platform/linux/CMakeLists.txt` :
```cmake
# set up ARM toolchain
set (CMAKE_C_COMPILER arm-linux-gnueabi-gcc)
set (CMAKE_AR  arm-linux-gnueabi-gcc-ar)
set (CMAKE_RANLIB arm-linux-gnueabi-ranlib)
```

and modify the `wamr_root/build-scripts/config_common.cmake` as following:
```cmake
62  # else ()
63      # add_definitions (-m32)
64      # set (CMAKE_EXE_LINKER_FLAGS "${CMAKE_EXE_LINKER_FLAGS} -m32")
65      # set (CMAKE_SHARED_LINKER_FLAGS "${CMAKE_SHARED_LINKER_FLAGS} -m32")
```
and try to build `iwasm` with `cmake .. -DWAMR_BUILD_TARGET=ARM` .  

Thanks.

yylittlefish, 2021-08-06:
@wustwn 
Hi,
When I use your method to build, I met this error:
[ 38%] Building ASM object CMakeFiles/vmlib.dir/home/yuyang/workspace/wasm/wasm-micro-runtime/wasm-micro-runtime/core/iwasm/common/arch/invokeNative_arm.s.o
/home/yuyang/workspace/wasm/wasm-micro-runtime/wasm-micro-runtime/core/iwasm/common/arch/invokeNative_arm.s: Assembler messages:
/home/yuyang/workspace/wasm/wasm-micro-runtime/wasm-micro-runtime/core/iwasm/common/arch/invokeNative_arm.s:24: error： no such instruction: `stmfd sp!,{r4,r5,r6,r7,lr}'
/home/yuyang/workspace/wasm/wasm-micro-runtime/wasm-micro-runtime/core/iwasm/common/arch/invokeNative_arm.s:25: error： too many memory references for `sub'
/home/yuyang/workspace/wasm/wasm-micro-runtime/wasm-micro-runtime/core/iwasm/common/arch/invokeNative_arm.s:26: error：  too many memory references for `mov'
/home/yuyang/workspace/wasm/wasm-micro-runtime/wasm-micro-runtime/core/iwasm/common/arch/invokeNative_arm.s:27: error：  too many memory references for `mov'
/home/yuyang/workspace/wasm/wasm-micro-runtime/wasm-micro-runtime/core/iwasm/common/arch/invokeNative_arm.s:28: error：  too many memory references for `mov'
/home/yuyang/workspace/wasm/wasm-micro-runtime/wasm-micro-runtime/core/iwasm/common/arch/invokeNative_arm.s:30: error：  expecting operand after ','; got nothing
/home/yuyang/workspace/wasm/wasm-micro-runtime/wasm-micro-runtime/core/iwasm/common/arch/invokeNative_arm.s:31: error：  no such instruction: `blt return'

do you know why?

BTW, I build it in another way, and it could generate iwasm successfully, I think this method is also okay, right?
1. I specify the toolchain.cmake in this way:
cmake .. -DCMAKE_TOOLCHAIN_FILE=~/wasm/wasm-micro-runtime/wasm-micro-runtime/samples/simple/profiles/arm-interp/toolchain.cmake -DWAMR_BUILD_PLATFORM=linux -DWAMR_BUILD_TARGET=ARMV7

2. modify product-mini/platforms/linux/CMakeList.txt, add this line:
 set (CMAKE_C_FLAGS "${CMAKE_C_FLAGS} -fPIC")
otherwise there's error:
tizen_cross_toolchain/tizen6.0_cross_toolchain_vd_target_armv7l_host_x86-64/bin/../lib/gcc/armv7l-tizen-linux-gnueabi/9.2.0/../../../../armv7l-tizen-linux-gnueabi/bin/ld: CMakeFiles/iwasm.dir/main.c.o: relocation R_ARM_MOVW_ABS_NC against `a local symbol' can not be used when making a shared object; recompile with -fPIC

wustwn, 2021-08-11:
> When I use your method to build, I met this error:
[ 38%] Building ASM object CMakeFiles/vmlib.dir/home/yuyang/workspace/wasm/wasm-micro-runtime/wasm-micro-runtime/core/iwasm/common/arch/invokeNative_arm.s.o
...

It should be because we use different cross compiler, compilation runs well when I use `arm-linux-gnueabi-gcc`  for target `ARM` and `ARM7V`.

> BTW, I build it in another way, and it could generate iwasm successfully, I think this method is also okay, right?

It looks normal, you can have a test with `iwasm` you built.

Thanks.

closed 2021-12-30 by wenyongh

arjunr2, 2022-02-15:
Hi @wustwn ,

I am also encountering this issue. I am using `arm-linux-gnueabi-gcc' to compile to ARMv7, and I get the same error as @yylittlefish. Is a specific compiler version required?

Thanks

wustwn, 2022-02-16:
Hey @arjunr2 

As I know, there is no specific requirement for the version of cross compiler `arm-linux-gnueabi-gcc`.

I tried to install cross compiler using following command on `ubuntu-20.04`: 

```shell

sudo apt-get install gcc-arm-linux-gnueabi g++-arm-linux-gnueabi 

```

then I can successfully build `iwasm`. I check the verison I installed, which is `arm-linux-gnueabi-gcc-9`, maybe you can have a try.

Thanks

wd-wy, 2023-02-16:
> @wustwn Hi, When I use your method to build, I met this error: [ 38%] Building ASM object [...]

> Hi @wustwn ,
> 
> I am also encountering this issue. I am using `arm-linux-gnueabi-gcc` to compile to ARMv7, and I get the same error as @yylittlefish. Is a specific compiler version required?
> 
> Thanks

You should arm-linux-guneabi-gcc to compile asm files, if not specify, it will use host default /usr/bin/cc to compile invokeNative_arm.s . 
For example( I use arm-linux-gnueabihf-gcc, you can also use arm-linux-gnueabi-gcc) , in your toolchain.cmake add 
SET(CMAKE_C_COMPILER ${toolchain_sdk_dir}arm-linux-gnueabihf-gcc)
SET(CMAKE_CXX_COMPILER ${toolchain_sdk_dir}arm-linux-gnueabihf-g++)
SET(CMAKE_AR ${toolchain_sdk_dir}arm-linux-gnueabihf-ar)
SET(CMAKE_RANLIB ${toolchain_sdk_dir}arm-linux-gnueabihf-ranlib)
SET(CMAKE_ASM_COMPILER ${toolchain_sdk_dir}arm-linux-gnueabihf-gcc)