wasm-micro-runtime/wasm-micro-runtime #688: Forward compatibility of generated AOT files

robinvanemden, 2021-08-05:
On compiling one of our WAMR applications using the latest WAMR head, it proved unable to run AOT files generated by earlier versions of the wamr-compiler. Will this remain the case for future commits of wamr/wamrc? That is, will it always be the case that a wamr app that makes use of AOT and the wamrc that generates its AOT files need to be compiled from (about) the same version wamr source code? Or is this intercompatibility issue mainly LLVM version related?

wenyongh, 2021-08-09:
@robinvanemden yes, the AOT file and runtime should match with each other, it is really not easy to run the older version of AOT file with latest runtime due to changes introduced by spec new features, e.g., spec proposal type import, typed reference function and exception handling, which introduce new contents and lead to the change of AOT file and AOT module instance layout.

robinvanemden, 2021-08-09:
Thanks for your answer, @wenyongh . That makes sense. We'll make sure to couple runtime & AOT versions accordingly!

closed 2021-08-09 by robinvanemden