wasm-micro-runtime/wasm-micro-runtime #699: How is wamr tested?

mbbill, 2021-08-18:
I went through the docs/wiki and couldn't find a way to run unit tests or integration tests.
So how it wamr tested against the standard? Is there a way to run wamr on the [test suite](https://github.com/WebAssembly/testsuite)?

How  are the wamr-specific features tested?  e.g. AOT 

My questions are mostly because I'd like to make sure I don't break anything after making  changes.

yamt, 2021-09-08:
> Is there a way to run wamr on the test suite?

i'm interested in the answer too.
the existence of WAMR_BUILD_SPEC_TEST option made me think there is a reasonably easy way to run these tests.
while it doesn't seem too difficult to write an equivalent of https://github.com/wasm3/wasm3/blob/main/test/run-spec-test.py by myself, if there is an existing tool, i want to know.

yamt, 2021-09-08:
@wenyongh hi. i suppose you can answer this question as you introduced WAMR_BUILD_SPEC_TEST.

xwang98, 2021-09-08:
There are some test suites used by the core team that are not released yet. We have plan to release it and will keep you updated. @mbbill  @yamt

yamt, 2021-09-08:
@xwang98 i see. thank you!

yamt, 2021-09-08:
@xwang98
can your non-released test suites run these tests? https://github.com/WebAssembly/spec/tree/master/test/core
if yes, which one pass and which ones fail?
asking because my colleague is trying to run them.

wenyongh, 2021-09-09:
@mbbill @yamt The WAMR test suites were uploaded, please ref to https://github.com/bytecodealliance/wasm-micro-runtime/tree/main/tests/wamr-test-suites for more details.

yamt, 2021-09-09:
@wenyongh thank you! i will take a look.

closed 2021-09-09 by mbbill