wasm-micro-runtime/wasm-micro-runtime #703: in arm64, when call a wasm function, get Exception: unreachable

victortanjh, 2021-08-23:
In the native world, calling wasm_runtime_call_wasm to call a wasm function, sometimes, I will get a exception:unreachable.

How can I get the call stack in wasm, and how about the reason of this exception?
Thank you.

victortanjh, 2021-08-24:
there is another exception: find block address failed.
I find this exception is related to compilation option:-DWAMR_BUILD_INTERP=1.
If I remove this compilation option, both exception:"unreachable" and "find block address failed" disappeared

xujuntwt95329, 2021-08-24:
Hi, for the first question, you can build with `WAMR_BUILD_CUSTOM_NAME_SECTION=1` and `WAMR_BUILD_DUMP_CALL_STACK=1` (refer to [build wamr.md](https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/doc/build_wamr.md#configure-debug)), then the call stack will be dumped once exception occurs. 

The "unreachable" exception is caused by the `unreachable opcode`, it is generated by the compiler to indicate that the code should never reach here, so it is hard to say what's the root cause from the runtime level, you may need to analyze the call stack to find out the abnormal code path.

For the second question, could you please provide the wasm file?

swankjesse, 2025-02-25:
I also ran into this with these compiler flags:

```
set(WAMR_BUILD_INTERP 1)
set(WAMR_BUILD_FAST_INTERP 0)
set(WAMR_BUILD_AOT 0)
```

I was able to work-around it with these compiler flags.

```
set(WAMR_BUILD_INTERP 1)
set(WAMR_BUILD_FAST_INTERP 1)
set(WAMR_BUILD_AOT 0)
```