wasm-micro-runtime/wasm-micro-runtime #705: pthread_cond_timedwait

wurc, 2021-08-24:
we use pthread_cond_timedwait as below, using wasi-sdk build is okay
#include "tizen_api.h"
#include <stdio.h>
#include <pthread.h>
pthread_mutex_t lock;
pthread_cond_t cond;

void *thread_routine(void *arg)
{
    printf("Enter thread\n");
    int ret = 0;
    int count = 0;
    while(count < 20)
    {
       count++;
       pthread_mutex_lock(&lock);
       pthread_cond_timedwait(&cond, &lock, 500000);
       pthread_mutex_unlock(&lock);

    }
    pthread_exit(NULL);
    return NULL;
}

int main(int argc, char **argv)
{
    pthread_mutex_init(&lock, NULL);
    pthread_cond_init(&cond, NULL);
    pthread_t tid;

    if (0 != pthread_create(&tid, NULL, thread_routine, NULL)) {
        printf("Failed to create thread\n");
    }

    if (0 != pthread_join(tid, NULL)) {
        printf("Failed to join thread %d.\n", tid);
    }
    pthread_cond_destroy(&cond);
    pthread_mutex_destroy(&lock);
    printf("Exit\n");

    return 0;
}
but when run pop up below error
[02:31:20:054 - B60FDE40]: failed to check signature '(**I)i' and resolve pointer params for import function (env pthread_cond_timedwait)
it means the parameter type mismatch ?

wurc, 2021-09-01:
we have use add "--sysroot=wamr-sdk/app/libc-builtin-sysroot/" when build
the third parameter is "I" so 500000 is not right?

xujuntwt95329, 2021-09-08:
That's strange, did you see any warning during compilation ?

You can use `wasm-objdump -x xxx.wasm` to check the actual signature of the `pthread_cond_timedwait` function

closed 2022-09-23 by wenyongh