wasm-micro-runtime/wasm-micro-runtime #735: Please provide more detail in embedding documentation

itowlson, 2021-09-09:
I am looking to embed the micro runtime into a host derived from another program's samples, but am having trouble following the documentation.  The sample is very minimal, and its makefile is just to `cc` three source files and then `ld` them together into an image.

I did a Linux build of the mini-product and got a `.so` out, but `ld` couldn't link that with my . giving an error about the symbols being in a different format.  I have tried adding `.c` files to the sample and to its makefile, but getting them to build is proving absurdly laborious because the existing makefile is so minimal, and the mini-product uses a very different build system (and the generated makefiles are beyond my skills).

Now this may be down to the minimalism of my target - I am trying to embed into a bare image rather than an ELF or PE format, and I don't have a stdlib - but with the current embedding documentation I can't tell if it's that or if I'm missing something!  It would be super useful if the embedding docs, as well as showing the APIs to call, provided information and examples of how to _build_ and _link_ the WAMR files outside of the WAMR mini product.

Thanks!

xujuntwt95329, 2021-09-10:
Hi, since there are so many building systems (e.g. make, cmake, autoconf, scons, gyp, etc), it is really hard to document the building process to cover all these tools. 

We use `cmake` for most of the components, and if you are using cmake, we have provided useful `cmake` scripts under [build-scripts](https://github.com/bytecodealliance/wasm-micro-runtime/tree/main/build-scripts)

If you are using `Makefile`, we also have some platform using `Makefile` as the building tools, please refer to [aos.mk](https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/product-mini/platforms/alios-things/aos.mk), this file is quite simple, all you need to do is to add the source files and include files into the `Makefile`

If you encounter any problem, could you please provide more details or attach the building logs here so that we can point out the possible solution?

Thanks!

dosgo, 2021-09-23:
If I want to embed it in the golang language to run the wasm file, do I refer to this document? https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/doc/embed_wamr.md Language binding of this library

xujuntwt95329, 2021-09-23:
@dosgo Yes, you can refer to that documents, and what you need to do is to wrapper those APIs and data structures in go.

Alternatively, you can also use wasm-c-api (refer [here](https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/doc/wasm_c_api.md)) to achieve your goal, the number of wasm-c-api is much more than WAMR's embedding API, but it may provide more flexibility.