wasm-micro-runtime/wasm-micro-runtime #741: Variable name `export` conflicting with C++11 keyword

csegarragonz, 2021-09-09:
Hi,

I am using WAMR in a C++ project, and the variable name `export` in [`core/iwasm/common/runtime_common.h`](https://github.com/bytecodealliance/wasm-micro-runtime/blob/db695fada42e6eed7cc9a94af7a48b4f1369009e/core/iwasm/common/wasm_runtime_common.h#L782) conflicts with the C++11 keyword.

xujuntwt95329, 2021-09-10:
Hi, thanks for reporting this, we'll fix this later.

As a quick workaround, you can use a macro to shadow `export` before including this file to make the compiler happy, for example:
``` cpp
extern "C" {
  /* shadow the C++ keyword "export" */
  #define export extern_export
  
  #include "../interpreter/wasm_runtime.h"
  
  #undef export
}
```

csegarragonz, 2021-09-10:
Yes, I was doing that before, thanks.

I have issues a very small PR to fix this.

PR #745 (2021-09-11): Change `export` variable name 
Otherwise it conflicts with a C++ keyword.

Closes #741
closed 2021-09-11 by wenyongh