wasm-micro-runtime/wasm-micro-runtime #744: wasm module can not pass parameters into other module

tkernelcn, 2021-09-10:
please refer to the following demo code,  I pass parameters to sub module,  **but sub modue print out It is 0**,  what's wrong with my usage method or some Issue in iwasm. 
(the simple demo can run on linux correctly that builded by GCC, but wasm not)
Thanks!

peter@ubuntu: ~/temp/test_wasm$ cat mod_add.c
#include <stdio.h>

int add(int a, int b)
{
    printf("in add, a=%d, b=%d\n", a,b);
    return a+b;
}
peter@ubuntu: ~/temp/test_wasm$ cat mod_add.h
#ifndef __MOD_ADD_H__
#define __MOD_ADD_H__

__attribute__((import_module("mod_add")))
__attribute__((import_name("add"))) extern int add(int a, int b);

#endif
peter@ubuntu: ~/temp/test_wasm$ cat main.c
#include <stdio.h>
#include "mod_add.h"

int main(int argn, char *argv[])
{
    printf("hello wasm world!, mod_add result=%d\n", add(20,32));
    return 0;
}

peter@ubuntu: ~/temp/test_wasm$ clang-8 --target=wasm32 -z stack-size=4096 -nostdlib -Wl,--no-entry,--allow-undefined,--export-all mod_add.c -o mod_add.wasm
peter@ubuntu: ~/temp/test_wasm$ clang-8 --target=wasm32 -O3 -z stack-size=4096 -Wl,--initial-memory=65536 -Wl,--allow-undefined,--export=main -Wl,--strip-all,--no-entry -nostdlib main.c -o hello.wasm
peter@ubuntu: ~/temp/test_wasm$ ./iwasm hello.wasm
in add, a=0, b=0
hello wasm world!, mod_add result=0
peter@ubuntu: ~/temp/test_wasm$

wenyongh, 2021-09-14:
@tkernelcn Thanks for reporting the issue! The issue was fixed by #748, could you please try again?

tkernelcn, 2021-09-14:
perfect! It works!

closed 2021-09-14 by tkernelcn

tkernelcn, 2021-09-14:
//append test results:
peter@ubuntu:~/temp/test_wasm$ ./iwasm hello.wasm
in add, a=20, b=32
hello wasm world!, mod_add result=52

tkernelcn, 2021-09-15:
@wenyongh  hello, one question to you
currently only support type of parameters are `int64_t` and `double`, not support `char *buf`, right?
will it be supported in future?
Thanks.