wasm-micro-runtime/wasm-micro-runtime #812: The first parameter sensor_t of sensor_event_handler_f is no use

LinJian1982, 2021-10-31:
what's the usage of this parameter?

wustwn, 2021-11-01:
Hi, the parameter `sensor_t sensor` of the `sensor_event_handler_f handler` indicates which sensor the callback event belongs to, so:
1. it can distinguish which sensor triggers the event. For example, we can open two sensors, and let them share a same `sensor_event_handler_f handler`, and use the `sensor_t sensor` argument to distinguish which sensor triggers the event.
2. we might need to get sensor attributes, e.g. name and interval, these APIs might be added in the future.

LinJian1982, 2021-11-01:
so this design encourage developer share the same event_call_back_function among different sensors, and make their code all together. 
while if each sensor has its own callback_function just make thing more easier.

closed 2021-12-13 by LinJian1982