wasm-micro-runtime/wasm-micro-runtime #816: Any good example for using the native API in iwasm?

gnh1201, 2021-11-04:
Can I use the native API of the operating system such as file input/output?

Is there any example or documentation of implementing this in iwasm?

xujuntwt95329, 2021-11-09:
Yes, you can write some wrapper APIs to expose native APIs to wasm. Here is the document for [exporting native APIs](https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/doc/export_native_api.md#export-native-api-to-wasm-application), you can follow this document to write your own APIs.

Also there is an example which demonstrates this usages:

- [basic](https://github.com/bytecodealliance/wasm-micro-runtime/tree/main/samples/basic): native APIs are implemented in [native_impl.c](https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/samples/basic/src/native_impl.c), these APIs are registered into runtime in [main.c](https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/samples/basic/src/main.c), and then we can invoke these APIs in [wasm application](https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/samples/basic/wasm-apps/testapp.c)

Hope this can help you!

xujuntwt95329, 2021-11-09:
BTW, if you just want to use some file I/O operations, you don't even need to write a native wrapper, you can use `WASI (WebAssembly System Interface)` to achieve this. 

Please refer to [build_wamr.md](https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/doc/build_wamr.md#configure-libc) for how to configure libc library during build iwasm (WASI is enabled by default on Linux platform). 
And refer to [buld_wasm_app.md](https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/doc/build_wasm_app.md#build-wasm-applications-with-wasi-sdk) for how to build wasm application with WASI support.

gnh1201, 2021-11-09:
Thank you so much. It's such good content. I'll give it a try.