wasm-micro-runtime/wasm-micro-runtime #830: Can I implement json API at runtime?

JavanZhu, 2021-11-11:
Hi, I am using AssemblyScript to develop some applications which running at WAMR runtime, but currently we encountered a performance bottleneck about handling JSON data with AssemblyScript 
JSON library(WAMR aot), which costs a lot of time in our case(compared to quickjs). 
Seems that the memory manager in AssemblyScript (GC etc) is one of reasons influence its performance. 

some performance data:
CPU: ARMv7
| Library name | parse() | toString() |
| -----        |  -----  | -------    |
| cJSON(C) native | < 1ms | 1ms       |
| AssemblyScript JSON | 10ms | 8ms       |

Can I implement JSON API internally in runtime? Any ideas about improving the AssemblyScript json performance are welcome, thanks in advance.

lum1n0us, 2021-11-12:
May I ask this what do you mean by *implementing JSON API internally*? Would you mind giving some details about your plan?

JavanZhu, 2021-11-12:
Hi @lum1n0us, Thank you for your reply。
I am trying to use the following method to improve the performance of AssemblyScript parsing json:
1. Built-in some json libraries implemented in c language in runtime, such as cJson
2. Provide some APIs of cJson, such as parse(), to AssemblyScript through "Export native API to WASM application"(WAMR)

But encountered some problems, such as:
1. Frequent API calls via ""Export native API to WASM application"" will also be time-consuming.

I have some ideas about this, but I don't know if it is feasible, so I want to ask experienced people for advice. For example, after parsing the json string locally through the C language, can it directly return an object that can be used directly by AssemblyScript? 
If I have expressed something that confuses you,
Say sorry first, please feel free to ask me questions. Thanks

lum1n0us, 2021-11-14:
We'd like to see your ideas on reducing time consumption of calling native APIs. 

In the example of "after parsing the json string locally through the C language, can it directly return an object that can be used directly by AssemblyScript", I think the point of it is always use the WASM linear memory. In native APIs, it is OK to do memory operations. The implementation of 'strdup_wrapper' might be a good example.

JavanZhu, 2021-11-15:
@lum1n0us Yes, the key point is use the WASM linear memory. For 'strdup_wrapper' function, I check out the source code in WAMR:
https://github.com/bytecodealliance/wasm-micro-runtime/blob/3ded9ece83776863e260548cced140d11cb62ab4/core/iwasm/libraries/libc-builtin/libc_builtin_wrapper.c#L553-L572

seems that we use it to copy string. And Do you know any information or code examples on how to create an AssemblyScript object in WAMR Runtime? thanks.

lum1n0us, 2021-11-15:
May I ask what is *an AssemlbyScript Object*?  Is it a shared buffer between WAMR and natvive?

JavanZhu, 2021-11-15:
@lum1n0us In AssemblyScript， create an AssemblyScript Object like:
```
class MyClass {
  propOne: i32;
  propTwo: i32;
}
let a = new MyClass();
```
[AssemblyScript reference](https://www.assemblyscript.org/status.html#object-literals).

lum1n0us, 2021-11-15:
guess you could malloc and initiate an object in native APIs and access its fields with predefined layout knowledge or 'get/set' native APIs.

JavanZhu, 2021-11-16:
> guess you could malloc and initiate an object in native APIs and access its fields with predefined layout knowledge or 'get/set' native APIs.

Ok, thank you for your help.

closed 2021-11-16 by JavanZhu