wasm-micro-runtime/wasm-micro-runtime #842: Build WAMR vmcore error On Win32

bxkai, 2021-11-22:
On Window，build winth Win32
-- Build Configurations:
     Build as target X86_32
     CMAKE_BUILD_TYPE Release
     WAMR Interpreter enabled
     WAMR AOT enabled
     WAMR JIT disabled
     Libc builtin enabled
     Libc WASI enabled with uvwasi implementation
     Fast interpreter enabled
     Multiple modules disabled
     SIMD enabled
     Reference types disabled
-- Configuring done
-- Generating done

The errors is below：
      wasm-micro-runtime\core\iwasm\aot\aot_loader.c(1827): error C2039: “extra_plt_data”: 不是“AOTModule”的成员
      wasm-micro-runtime\core\iwasm\aot\aot_loader.c(1850): error C2039: “xmm_plt_count”: 不是“AOTModule”的成员

wenyongh, 2021-11-22:
@bxkai could you test whether #843 fixes the issue?

bxkai, 2021-11-23:
@wenyongh build win32 sucess.  But aot run error.

1) wamrc.exe --size-level=1 --target=i386 --target-abi=msvc --disable-simd -o test-i386-msvc test.wasm
Create AoT compiler with:
  target:        i386
  target cpu:
  cpu features:
  opt level:     3
  size level:    1
  output format: AoT file
Compile success, file test-i386-msvc was generated.

2)  iwasm test-i386-msvc
AOT module load failed: invalid target endian type, expected little endian but got big endian

wenyongh, 2021-11-29:
@bxkai seems that the win32 system is little endian while the aot file generated is big endian, no idea about that, could you try modifying aot_loader.c to ignore the check and see whether it can run success?

wenyongh, 2021-11-30:
@bxkai could you please test whether #855 fixes the issue?

bxkai, 2021-12-11:
@wenyongh 
iwasm.exe test-i386-msvc
AOT module load failed: invalid function section size

wenyongh, 2021-12-12:
Seems that endian/bit-width check in win32 is invalid, could you please whether patch #886 works?

bxkai, 2021-12-13:
@wenyongh 
iwasm.exe  test-i386-msvc
AOT module load failed: resolve symbol _aot_set_exception_with_id failed

wenyongh, 2021-12-13:
@bxkai I uploaded a [patch set](https://github.com/bytecodealliance/wasm-micro-runtime/pull/886/commits/842694aaa5ffd31d01a80de61e66277dda3ab477) to fix it, could you please try again?

bxkai, 2021-12-17:
@wenyongh 
Load relocation section failed: invalid relocation type 20

wenyongh, 2021-12-20:
Hi, do you run iwasm aot on win32 x86_32 target? The relocation operations for Windows x86_32 haven't been implemented yet. Could you please refer to the relocation handlings for Windows x86_64 on [core/iwasm/aot/arch/aot_reloc_x86_64.c](https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/core/iwasm/aot/arch/aot_reloc_x86_64.c#L123-L126) to implement them in file aot_reloc_x86_32.c?
Note that Windows object file format is PE format, refer to the following link for more details:
https://docs.microsoft.com/en-us/windows/win32/debug/pe-format#intel-386-processors
https://docs.microsoft.com/en-us/windows/win32/debug/pe-format#arm64-processors

For example, the relocation type 20 on x86_32 is IMAGE_REL_I386_REL32, the relocation operation might be similar to IMAGE_REL_AMD64_REL32 on x86_64.

bxkai, 2021-12-23:
win32 app is run on windows 64 system, I build iwasm win32 testing on win64 system.
I will try with your proposal.