wasm-micro-runtime/wasm-micro-runtime #852: Error when use "invokeNatvie_general.c"

iNoak, 2021-11-25:
How to use the C code version to call the native function instead of asm version?? 
I've tried to set WAMR_BUILD_INVOKE_NATIVE_GENERAL, but error occured.

wenyongh, 2021-11-29:
Hi, currently invokeNative_general.c is just for experiment purpose, normally it only works in some 32-bit target, e.g. x86-32:
https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/core/iwasm/common/iwasm_common.cmake#L17-L27
please use the assembly version instead.

iNoak, 2021-12-14:
Thank you.
As I'm not quite familiar with the runtime technics, I just wonder why I can't use the  "function pointer" in C language to call the native function (simulating the data movement in stack concurrently, by coping the parameters and rerturn value).
Why is necessary to use the assembly code?
Thanks for patient reply!

wenyongh, 2021-12-21:
Hi, the main reason to use the assembly code to call the native function pointer (includes AOT func ptr) is to improve the user experience, by this way, we can define the native API with the exact prototype as it should be, for example, we can define native API of signature "(ii)i" as `int foo(int, int)`, and define natvie API of signature "(*~)i" as `int foo(uint8* buf, uint32 buf_length)`, it should be very friendly to the developer.

And if we don't use asm code, normally we need to use a uint64 array to pass the parameters from wasm app to native, and get back the return value from native to wasm, e.g.
```C
void foo(uint64 *argv, int argc) {
    param1 = (param1_type)argv[0];
    param2 = (param2_type)argv[1];
    ...
    argv[0] = (return_type)ret_value;
}
```
It is not so convenient to developer.
In fact  WAMR supports to register native API as raw mode, in which developer can define the native API like above, please refer to:
https://github.com/bytecodealliance/wasm-micro-runtime/issues/220#issuecomment-607061112
https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/core/iwasm/include/wasm_export.h#L768-L781
for more details.

In this mode runtime doesn't call the asm code to invoke the native APIs registered as raw mode. The issue is that the builtin native libraries (libc-builtin, libc-wasi, lib-pthread, and so on) and AOT code are still called by asm code. So you might not be able to run wasm app in AOT mode, and if you want to use the builtin native libraries, you might need to re-implement them by yourself.

The other reason is for performance, in our test, the performance of asm code mode is better than raw mode. Thanks.