wasm-micro-runtime/wasm-micro-runtime #854: Questions about "-sgx" option when generating AoT file for SGX platform

qinkunbao, 2021-11-29:
I was trying to compile (`wamrc  -sgx -o app.aot app.wasm `) an AoT file from a wasm file. I found the size of the generated AoT file `app.aot` is significantly larger than the original wasm file.  Then I noticed that the code size level for an SGX mode is only 1. 

https://github.com/bytecodealliance/wasm-micro-runtime/blob/9b0b33e4aedf4324b91bbf402ce62ee00c44ef5e/wamr-compiler/main.c#L211

As the generated AoT file needs to be loaded into the SGX EPC memory and the memory resource is limited (96 MB on SGX1), the file size of a generated AOT file for an SGX mode does matter a lot in many situations. I was wondering why the code size level is only 1 (The highest level is 3) when generating an AoT file for the Intel SGX platform. Thanks.

wenyongh, 2021-11-30:
Hi, to run the AOT code, normally the runtime should allocate the executable memory and apply relocations by patching the allocated memory, as some relocation types generated by LLVM codgen with size level 3 require that the target addresses (e.g. some place of the allocated aot code) must be in range 0-2GB. But for SGX, normally the address of memory allocated is larger than 4G, so we have to set size level to 1 to yield these relocation types.

There might be another solution for this, currently WAMR has implemented the XIP (Execution In place) feature to run the AOT file directly, without patching to it (or do few patches), could you please try this patch:
[sgx_xip.zip](https://github.com/bytecodealliance/wasm-micro-runtime/files/7622417/sgx_xip.zip)
and for wamrc
```bash
wamrc --enable-indirect-mode --disable-llvm-intrinsics -sgx -o <aot_file> <wasm_file>
```
for iwasm sgx:
```bash
iwasm --xip
```
Thanks.

qinkunbao, 2021-11-30:
Thank you for your prompt reply.

> Hi, to run the AOT code, normally the runtime should allocate the executable memory and apply relocations by patching the allocated memory, as some relocation types generated by LLVM codgen with [...]

That makes sense to me. Can I compile a wasm file with the size level 2 option? I roughly remember that the recent Intel SGX SDK and driver allow an application to load an enclave image at a user-defined virtual address (<2GB). It is a new feature that is designed to prevent the NULL pointer deference attack. I will do a few experiments and get back to you.

> There might be another solution for this, currently WAMR has implemented the XIP (Execution In place) feature to run the AOT file directly, without patching to it (or do few patches), could you please try this patch: [sgx_xip.zip](https://github.com/bytecodealliance/wasm-micro-runtime/files/7622417/sgx_xip.zip) and for wamrc
> 
> ```shell
> wamrc --enable-indirect-mode --disable-llvm-intrinsics -sgx -o <aot_file> <wasm_file>
> ```
> 
> for iwasm sgx:
> 
> ```shell
> iwasm --xip <wasm_file>
> ```
> 
> Thanks.

I am sorry, but I can't find any documents about the XIP mode. Does the XIP mode finish the relocation during the compiling time?

wenyongh, 2021-11-30:
> Thank you for your prompt reply.
>
>
> That makes sense to me. Can I compile a wasm file with the size level 2 option? I roughly remember that the recent Intel SGX SDK and driver allow an application to [...]

Compiling with size level 2 might also generate unexpected relocation types. It would be great if the latest SGX SDK supports   loading the enclave image into 0-2GB, you can test whether SGX API sgx_alloc_rsrv_mem(length)/sgx_alloc_rsrv_mem_ex(desired_addr, length) returns address in 0-2GB, if yes, then we can use --size-level=2/3 for wamrc.

> I am sorry, but I can't find any documents about the XIP mode. Does the XIP mode finish the relocation during the compiling time?

Sorry, currently XIP document hasn't been added, we will update soon. XIP mode tries to emit code without relocation records, so that aot loader doesn't need to patch the aot code, the goal is to run aot file from flash without duplicating/modifying the aot code. Currently in most cases no relocations are required, but in some cases some relocations to ".text"/".rodata" are generated, we will try to fix that in the future.

wenyongh, 2021-12-06:
@qinkunbao we have merged the patch #863, the XIP file generated by `wamrc --enable-indirect-mode --disable-llvm-intrinsics -sgx` should be able to run on SGX platform, could you please try again?

BTW: have you tested the latest Intel SGX SDK to check whether it can load the enclave image into a user-defined virtual address which is smaller than 2GB? Thanks.

qinkunbao, 2021-12-08:
> @qinkunbao we have merged the patch #863, the XIP file generated by `wamrc --enable-indirect-mode --disable-llvm-intrinsics -sgx` should be able to run on SGX platform, could you please try again? [...]

Here is what I got.
WAMR Version: commit b9fd8f916b3dc71ecff0fb0aef00fee384e898f5

```bash
./wamrc --enable-indirect-mode --disable-llvm-intrinsics -sgx -o test.aot test.wasm 
```

I used the `sgx iwasm` under the folder [encalve_sample](https://github.com/bytecodealliance/wasm-micro-runtime/tree/main/product-mini/platforms/linux-sgx/enclave-sample).

```bash
./iwasm --dir=. ./test.aot

Segmentation fault (core dumped)
```

Then I tested the same AOT file under the `iwasm` under the folder 
[iwasm](https://github.com/bytecodealliance/wasm-micro-runtime/tree/main/product-mini/platforms/linux). The program was running smoothly and I got the expected result.

Then I recompiled the wasm program with the following command
```bash
./wamrc -sgx -o test.aot test.wasm
```
The AOT files worked well with the `iwasm` under the folder [linux](https://github.com/bytecodealliance/wasm-micro-runtime/tree/main/product-mini/platforms/linux). However, I got an exception `EXCE_OUT_OF_BOUNDS_MEMORY_ACCESS` from the sgx iwasm under the folder [encalve-sample](https://github.com/bytecodealliance/wasm-micro-runtime/tree/main/product-mini/platforms/linux-sgx/enclave-sample).

qinkunbao, 2021-12-08:
> @qinkunbao we have merged the patch #863, the XIP file generated by `wamrc --enable-indirect-mode --disable-llvm-intrinsics -sgx` should be able to run on SGX platform, could you please try again? [...]

> 

I don't have access to the machine with the latest Intel SGX driver at the present. I will let you know when I have any updates.

Here are the relevant documents, in case you are interested.

[Intel SGX SDK](https://download.01.org/intel-sgx/sgx-linux/2.15.1/docs/Intel_SGX_Developer_Reference_Linux_2.15.1_Open_Source.pdf) Please search the key word `ELRangeStartAddress`.

[Open Encalve](https://github.com/openenclave/openenclave/blob/master/docs/DesignDocs/SGX_0-base_Enclave.md)

wenyongh, 2021-12-08:
@qinkunbao we have submitted patch to fix it, could you please try again? Note that both wamrc and iwasm for sgx should be rebuilt.

qinkunbao, 2021-12-21:
It works for me now. Thanks.

closed 2022-02-17 by wenyongh