wasm-micro-runtime/wasm-micro-runtime #859: Multi-Module and WASI

JHNX, 2021-12-01:
Dear there,

As shown in wasm_application_execute_main(), in wasi mode, we should call function named "_start" firstly to trigger the real main().
I tried to build one .wasm which includes callings to C++ library, if execute with "./iwasm --function ABC abc.wasm", indeed exception is encountered. BTW, if the wasm implementation doesn't include callings to C++ library, there's no issue.

Looks like every .wasm shall be entered from the single entry "_start". Now I want to enable multi-module, in one module, how to call functions of other modules?

Thanks in advance.

lum1n0us, 2021-12-01:
Here is an [example](https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/samples/multi-module/src/main.c#L119) of three modules, A, B, and C.  

Let's say C is the main module. It depends on the functions of B and A. We only need to call `wasm_runtime_load` with *C* once and WAMR will load *A* and *B* automatically. After that, we can use `wasm_application_execute_func` with a string like *$MODULE_NAME$FUNCTION_NAME* to indicate any function in A ,B, or C .

lum1n0us, 2021-12-01:
BTW, neither of them has a function named *main* or *_start*

JHNX, 2021-12-01:
Thanks for your quick response. Yes, the usage of multi-module is clear as shown in samples/multi-module. But the issue is if the .c file is changed to .cpp file, build them with wasi-sdk and add function calling c++ library, such as calling "cout" to dump something, then call the function from the other module, it cannot run as expected that exception is encountered. 
BTW, if DO NOT call "cout", it works well.

lum1n0us, 2021-12-02:
Nice catch.

lum1n0us, 2021-12-20:
@JHNX  Thanks a lot!

[Reset the register stack and counter](https://github.com/WebAssembly/wasi-libc/pull/264)

lum1n0us, 2021-12-30:
We should follow the WASI command/reactor model here. Please refer to our [sample of multiple modules](https://github.com/bytecodealliance/wasm-micro-runtime/tree/main/samples/multi-module).

In a nutshell, 
- The main module is a command in the WASI model. It imports items from its sud-modules (reactors in the WASI model)
- The submodule likes a library exports items. To be a reactor, need to be compiled with `-mexec-model=reactor`