wasm-micro-runtime/wasm-micro-runtime #861: Error while using source debug: "Process 1 stopped and restarted: thread 1 received signal: SIGCONT"

JavanZhu, 2021-12-02:
Hi, all. I tried the source debug according to the [tutorial](https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/doc/source_debugging.md). 
...
This command compiled an error, so I added a compilation option `-DLLDB_INCLUDE_TESTS=OFF` to fix it.

```
cmake -DCMAKE_BUILD_TYPE:STRING="Release" -DLLVM_ENABLE_PROJECTS="clang;lldb" -DLLVM_TARGETS_TO_BUILD:STRING="X86;WebAssembly" -DLLVM_ENABLE_LIBXML2:BOOL=ON ../llvm
```

But after running the following command to debug:
```
 process connect -p wasm connect://127.0.0.1:1234
```
 I got the following output: 
lldb log ->
```
(lldb) process connect -p wasm connect://127.0.0.1:1234                                                                                                                                                                                 
Process 1 stopped and restarted: thread 1 received signal: SIGCONT
(lldb) r
There is a running process, kill it and restart?: [Y/n] n
(lldb) b main
warning: failed to set breakpoint site at 0x40000000000000b1 for breakpoint 1.1: error sending the breakpoint request
Process 1 exited with status = -1 (0xffffffff) lost connection
Breakpoint 1: where = nobody.wasm`main + 55 at hello_world.c:6:5, address = 0x40000000000000b1

```
iwasm output log ->
```
...
[05:38:55:164 - 10188000]: send replay:0016046e616d65010f0200067072696e7466016d61696e000085010970726f64756365727302086c616e67756167650103433939000c70726f6365737365642d62790105636c616e675631302e302e30202868747470733a2f2f6769746875622e636f6d2f6c6c766d2f6c6c766d2d70726f6a656374206433323137306462643562306435343433363533376236623735626561663434333234653063323829
[05:38:55:348 - 10188000]: receive request:m 40000000000003b5,360

[05:38:55:401 - 10188000]: wasm_debug_instance_get_data_mem size over flow!

[05:38:55:469 - 10188000]: send replay:
[05:38:55:629 - 10188000]: receive request:m 400000000000032d,400

[05:38:55:664 - 10188000]: wasm_debug_instance_get_data_mem size over flow!

[05:38:55:688 - 10188000]: send replay:0085010970726f64756365727302086c616e67756167650103433939000c70726f6365737365642d62790105636c616e675631302e302e30202868747470733a2f2f6769746875622e636f6d2f6c6c766d2f6c6c766d2d70726f6a656374206433323137306462643562306435343433363533376236623735626561663434333234653063323829
[05:38:55:820 - 10188000]: receive request:m 40000000000003b5,378

[05:38:55:852 - 10188000]: wasm_debug_instance_get_data_mem size over flow!

[05:38:55:882 - 10188000]: send replay:
[05:38:55:990 - 10188000]: receive request:m 40000000000003b5,400

[05:38:55:019 - 10188000]: wasm_debug_instance_get_data_mem size over flow!

[05:38:55:041 - 10188000]: send replay:
[05:38:55:496 - 10188000]: receive request:q Symbol::

[05:38:55:531 - 10188000]: handle_generay_query:Symbol

[05:38:55:546 - 10188000]: send replay:
[05:38:55:144 - 10188000]: receive request:v Cont;C13:115c3be00

[05:38:55:216 - 10188000]: handle_v_packet:Cont

[05:43:46:413 - 10188000]: receive request:Z 0,40000000000000b1,1

[05:43:46:511 - 10188000]: send replay:OK
[05:43:46:561 - 10188000]: Connection closed
[05:43:46:576 - 10188000]: control thread of debug object 0x7ffdf0d05b70 stop

```

Not sure if it’s something wrong or it’s just that I use it incorrectly

----

environment:
* platform: mac Big Sur 11.2.3
* c source code:
```
#include <stdio.h>

int
main(int argc, char **argv)
{
    printf("hello webassembly\n");
    return 0;
}
```

xujuntwt95329, 2021-12-06:
Hi, seems the signal numbers on Mac are different from Linux, could you please add `platform select remote-linux` before connecting to WAMR?

``` bash
(lldb) platform select remote-linux
(lldb) process connect -p wasm connect://127.0.0.1:1234
```

I'll find a device to test it and update the document later.

JavanZhu, 2021-12-06:
That works, thank you.

closed 2021-12-06 by JavanZhu