wasm-micro-runtime/wasm-micro-runtime #864: Non-pointer paramater types in function signatures return incorrect values on armv7

robinvanemden, 2021-12-04:
Non-pointer paramater types for function signatures in NativeSymbols return incorrect values for our armv7 toolchain.

They work fine on x64 and aarch64, however.

For instance, the following function declaration:

```
void leaky_relu_accel(wasm_exec_env_t exec_env,
                      float *X,
                      int x_len,
                      float *Y,
                      int y_len,
                      float alpha);
```

... fails on armv7 when using: 

```
"leaky_relu_accel",
leaky_relu_accel,
"(*~*~f)"
```

 ... we can however work around this issue by using pointer-to-float-notation:

```
"leaky_relu_accel",
leaky_relu_accel,
"(*~*~*~)"
```

It would, however, be preferable to able to use the `i`, `I`, `f` and `F` function signatures accross the board.

wenyongh, 2021-12-06:
@robinvanemden does the target arch support  VFP coprocessor registers? Could you please try adding `_VFP` for WAMR_BUILD_TARGET when building iwasm? e.g. `cmake .. -DWAMR_BUILD_TARGET=ARMV7_VFP`, see:
https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/doc/build_wamr.md#configure-platform-and-architecture

robinvanemden, 2021-12-06:
@wenyongh changing our WAMR_BUILD_TARGET from `-DWAMR_BUILD_TARGET=ARMV7_VFP` to `-DWAMR_BUILD_TARGET=ARMV7_VFP` did indeed resolve this issue. Thanks! 

We do prefer to compile a single AOT that can be run on as broad an array of ARMV7 processors as possible. So we might just leave the [_VFP] suffix out and keep the "pointer-to-float-notation" for our function signatures. Benchmarks seem to show no performance differences between both options. Do you foresee any other issues we might encounter by going the "pointer-to-float-notation"-way for now?

wenyongh, 2021-12-07:
Hi, if there are float or double types (not float/double pointers) in the native function's parameters, please do add the _VFP suffix. And it should be OK to skip _VFP if there is no float/double type in the native function's paraments.
It seems a little tricky to use `"(*~*~*~)"`, maybe you can define the native function as:
```C
void leaky_relu_accel(wasm_exec_env_t exec_env,
                      float *X,
                      int x_len,
                      float *Y,
                      int y_len,
                      int32 i32);
{
    float alpha = *(float *)&i32; //or use memcpy
    ...
}
```
And register it with "(*~*~i)", and in wasm app, change float to i32 before calling the native func.

robinvanemden, 2021-12-07:
@wenyongh We'll adapt our code according to your suggestions - thanks for the clear explanation!

closed 2021-12-07 by robinvanemden