wasm-micro-runtime/wasm-micro-runtime #867: Invalid relocation type 282 for AOT on aarch64

robinvanemden, 2021-12-05:
On compiling to WASM using

```
WASM_FLAGS = --sysroot=${WASM_SYSROOT} -O3 -fno-exceptions -mbulk-memory -m32 -fno-rtti -ffast-math -fno-math-errno
```
and then compiling AOT with either wamrc_v0 (May 2021) and wamrc_v1 (current version, 12/5/2021):

```
wamrc_v0 --target=aarch64 --enable-bulk-memory -o model.aarch64_0.aot mode.wasm
wamrc_v1 --target=aarch64                      -o model.aarch64_1.aot model.wasm
```
we receive the following error when running the latest model.aarch64_1
```
Error: Load wasm runtime_core.module failed. error: Load relocation section failed: invalid relocation type 282.
```
The older wamrc_v0 does not generate this error. Nor does compiling AOT for targets x64 and armv7.

wenyongh, 2021-12-06:
@robinvanemden relocation type 282 is R_AARCH64_JUMP26, the relocation for it should be same as for R_AARCH64_CALL26, could you please try whether patch #873 works?

robinvanemden, 2021-12-06:
@wenyongh yes, patch #873 resolves this issue. Thanks!

closed 2021-12-07 by robinvanemden

Jiax-cn, 2025-02-19:
I encountered a similar issue: `Load relocation section failed: invalid relocation type 261.` Is there any plan to support this relocation type? Thank you very much!

lum1n0us, 2025-03-12:
To my knowledge, there isn't a plan in place for *data relocation*(including `R_AARCH64_ABSxxx` and `R_AARCH64_PRELxxx`) as seen in the Linux source (https://elixir.bootlin.com/linux/v4.9.12/source/arch/arm64/kernel/module.c#L232). However, you are welcome to contribute to this area.

yamt, 2025-04-03:
> [@wenyongh](https://github.com/wenyongh) I encountered a similar issue: `Load relocation section failed: invalid relocation type 261.` Is there any plan to support this relocation type? Thank you very much!

i guess you can file a new bug with a recipe to reproduce.