wasm-micro-runtime/wasm-micro-runtime #868: Concerning mutil-module and "sub_module_instantiate"

iNoak, 2021-12-06:
Hi there!
When I tried to use the "multi-modules" feature ( the parent module attempts to import a function from the sub module) , it seems like in the  instantiation step( "sub_module_instantiate" function ),  the parent module will instantiate the WHOLE sub module. 
Here is the question:  if sub module is a utils module containing enormous functions, the parent moudle just need one function from it. If the sub module is another app, the parent module just need very little from it's export segment (like a global). Will it  cause high overhead to instantiate every sub module in the parent module's sub_module_list?
Thank you.

iNoak, 2021-12-06:
To be brief, Why not let each module be instantiated only once？（sub module instance can be shared among several parent module).Is this a standard process or for some security purposes?
Thank you.

lum1n0us, 2021-12-07:
Yes. It is for security.

It is called a *shared-nothing* linking in WASM world. It tends to link multiple modules without exposing anyone's linear memory. Every instance works in its own sandbox and passes values only.  

Every parent module instance will have its own sub-module instance. It prevents a malicious parent module instance from peeking other parent module instances' sensitive data via shared sub-module instances.

iNoak, 2021-12-11:
Thank you for the reply.
So if there are two wasm apps ( A and B) running in the WAMR, A imports a global from B. 
If B change that global value in the runtime, how can B pass that value to A ?
I'm concerning about how to share dynamic data between two modules when running. Thank you.

lum1n0us, 2021-12-30:
I suppose you are talking about mimicking to share a global variable directly between two C/C++ files and we all agree to map the C global variable into a WASM global. In this special circumstance, WAMR will share the data of the global between A and B. Every read/writes from one module will be aware by another.

But don't forget another way by `set()/get()` functions. It is more elegant and easier to maintain.