wasm-micro-runtime/wasm-micro-runtime #877: mipsel AOT support

duchengxian, 2021-12-07:
2 problems:
1. wamrc support mips and mipseb, but mips==mipseb in wamrc. change source code to "mipsel" solved this problem
2. after above modification, iwasm can load AOT file, but reported "invalid relocation type", looks like mips doesn't do relocation

could you pls have a look ?

wenyongh, 2021-12-12:
Hi,

> 2 problems:
> 
> 1. wamrc support mips and mipseb, but mips==mipseb in wamrc. change source code to "mipsel" solved this problem

#887 adds "mipsel" to wamrc's target list, could you please try again with "wamrc --target=mipsel" after applying the patch?

> 2. after above modification, iwasm can load AOT file, but reported "invalid relocation type", looks like mips doesn't do relocation

Yes, currently the AOT relocation operations for mips target haven't been implemented, it really takes effort to resolve it. Could you please try the XIP mode instead? Please refer to the document [here](https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/doc/xip.md).

duchengxian, 2021-12-13:
#887 looks fine

I used XIP mode to generated AOT file, and iwasm has recognised the XIP mode, but apply_relocation() still be called and cause "Load relocation section failed: invalid relocation type 4." pls take a look if you have a chance

no1wudi, 2021-12-13:
@duchengxian 
Hi, now the XIP mode is highly architecture dependent feature, it's only tested on arm & riscv. Other platform would need to check the intrinsic need be replaced according to target ISA like this (float operation is easy to cause relocation usually):
https://github.com/bytecodealliance/wasm-micro-runtime/blob/5547924e28c589aee62210ef464833c659950328/core/iwasm/aot/aot_intrinsic.c#L602

duchengxian, 2021-12-13:
@no1wudi 
I am not familar with ISA, it's too difficult to find out how to adapt the runtime to MIPS

After comment out apply_relocation(), iwasm+aot is still not work, but it works after specific "-f". Maybe there are potential problems with long jump, but it's a workaround for short code

hope better solution for it. thanks for your great work

no1wudi, 2021-12-14:
What's your aot compile option ? 
Note: Simply comment out apply_relocation is dangerous , the relocation is occurred with some specific instructions instead of "long code".

duchengxian, 2021-12-15:
used option like wenyongh said:
wamrc --enable-indirect-mode --disable-llvm-intrinsics --target=mipsel -o <aot_file> <wasm_file>