wasm-micro-runtime/wasm-micro-runtime #878: Use WASI SDK to build wasm app

LinJian1982, 2021-12-08:
I'm trying to build the wasm app using the wasi sdk. The build is successful, but while I'm trying to install the app, it report following issue.
[16:28:16:621 - A39D8280]: warning: failed to link import function (env, strlen)
[16:28:16:821 - A39D8280]: warning: failed to link import function (env, malloc)
[16:28:16:826 - A39D8280]: warning: failed to link import function (env, puts)
[16:28:16:830 - A39D8280]: warning: failed to link import function (env, memcpy)
[16:28:16:833 - A39D8280]: warning: failed to link import function (env, free)
[16:28:16:837 - A39D8280]: warning: failed to link import function (env, memcmp)
[16:28:16:840 - A39D8280]: warning: failed to link import function (env, printf)
[16:28:16:843 - A39D8280]: warning: failed to link import function (env, strdup)
[16:28:16:846 - A39D8280]: warning: failed to link import function (env, strcmp)

I'm using the simple project and WAMR_BUILD_LIBC_WASI is defined in my profile. The build command is changed a little bit and highlighted as following:
/opt/wasi-sdk/bin/clang                                              \
        **-D__wasi__ -D__wasm32__**                                       \
        -I${WAMR_DIR}/wamr-sdk/out/$PROFILE/app-sdk/wamr-app-framework/include  \
        -L${WAMR_DIR}/wamr-sdk/out/$PROFILE/app-sdk/wamr-app-framework/lib      \
        -lapp_framework                                              \
        --target=wasm32 -O3 -z stack-size=4096 -Wl,--initial-memory=65536 \
        --sysroot=${WAMR_DIR}/wamr-sdk/out/$PROFILE/app-sdk/**wasi-sysroot**  \
        -Wl,--allow-undefined-file=${WAMR_DIR}/wamr-sdk/app/libc-builtin-sysroot/share/defined-symbols.txt \
        -Wl,--strip-all,--no-entry -nostdlib \
        -Wl,--export=on_init -Wl,--export=on_destroy \
        -Wl,--export=on_request -Wl,--export=on_response \
        -Wl,--export=on_sensor_event -Wl,--export=on_timer_callback \
        -Wl,--export=on_connection_data -Wl,--export=on_mqtt_event \
        -Wl,--export=__heap_base -Wl,--export=__data_end \
        -o ${OUT_DIR}/wasm-apps/${OUT_FILE} ${APP_SRC}

xujuntwt95329, 2021-12-08:
Hi, you use `-nostdlib` in your build command, this means don't link WASI library, so you are using `BUILTIN_LIBC`

If you do want to use WASI mode, remove the `-nostdlib` in your build command and compile again

If you want to use BUILTIN_LIBC, you need to enable `WAMR_BUILD_LIBC_BUILTIN`.

For the difference between WASI and BUILTIN_LIBC, please refer to this [document](https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/doc/build_wasm_app.md#1-wasi-sdk-options)

LinJian1982, 2021-12-09:
thanks for the quick feedback!
After remove the -nostdlib and change the --target==wasm32-wasi, I meet the new build issue.

The new build command as following:
/opt/wasi-sdk/bin/clang                                                 \
        -I${WAMR_DIR}/wamr-sdk/out/$PROFILE/app-sdk/wamr-app-framework/include \
        -L${WAMR_DIR}/wamr-sdk/out/$PROFILE/app-sdk/wamr-app-framework/lib     \
        -lapp_framework                                              \
        --target=**wasm32-wasi** -O3 -z stack-size=4096 -Wl,--initial-memory=65536 \
        --sysroot=${WAMR_DIR}/wamr-sdk/out/$PROFILE/app-sdk/wasi-sysroot  \
        -Wl,--allow-undefined-file=${WAMR_DIR}/wamr-sdk/app/libc-builtin-sysroot/share/defined-symbols.txt \
        -Wl,--strip-all,--no-entry \
        -Wl,--export=on_init -Wl,--export=on_destroy \
        -Wl,--export=on_request -Wl,--export=on_response \
        -Wl,--export=on_sensor_event -Wl,--export=on_timer_callback \
        -Wl,--export=on_connection_data -Wl,--export=on_mqtt_event \
        -Wl,--export=__heap_base -Wl,--export=__data_end \
        -o ${OUT_DIR}/wasm-apps/${OUT_FILE} ${APP_SRC}

my build issue as following:
#####################build wasm apps
wasm-ld: error: /root/wasm-micro-runtime/samples/simple/../../wamr-sdk/out/simple-host-interp/app-sdk/wasi-sysroot/lib/wasm32-wasi/libc.a(__main_argc_argv.o): undefined symbol: main
clang-11: error: linker command failed with exit code 1 (use -v to see invocation)
build algorithm.wasm fail
wasm-ld: error: /root/wasm-micro-runtime/samples/simple/../../wamr-sdk/out/simple-host-interp/app-sdk/wasi-sysroot/lib/wasm32-wasi/libc.a(__main_argc_argv.o): undefined symbol: main
clang-11: error: linker command failed with exit code 1 (use -v to see invocation)
build connection.wasm fail

LinJian1982, 2021-12-14:
@xujuntwt95329 any comments on this? basically I want to build the simple project with wasi SDK, and I'm trying to write app with C++

xujuntwt95329, 2021-12-14:
Hi @LinJian1982, sorry for the late reply as I didn't get a notification.

In `WASI` mode the main function is required, you can write an empty `main` in your app to solve this problem.

``` C
/* Implement empry main function as wasi start function calls it */
int
main(int argc, char **argv)
{
    (void)argc;
    (void)argv;
    return 0;
}

```

LinJian1982, 2021-12-14:
@xujuntwt95329  Thanks for the feedback.

But I would like to use the application framework, with which the app entry function is on_init. Should I modify the application framework to support main function?

xujuntwt95329, 2021-12-14:
@LinJian1982 The empty main function is just used to make the compiler happy in WASI mode, the app entry function will still be `on_init`. The main function will not be invoked if you are using the application framework.

LinJian1982, 2021-12-14:
@xujuntwt95329
Yeah, this is a good idea. let's me try it to make the compiler happy. :)

closed 2021-12-16 by LinJian1982