wasm-micro-runtime/wasm-micro-runtime #884: Potential sensor framework regarding the timer

LinJian1982, 2021-12-11:
It looks like we shall reset the ms_to_next_check if it is **large** than "s->read_interval". 
for example, we have two sensors in the system. The read interval of them are 100ms and 1000ms. The current code causes the ms_to_next_check be set to 1000ms, so the callback of the shorter one only be called after 1000ms.

if (elpased_ms >= s->read_interval) {
            attr_container_t * data = s->read(s);
            if (data) {
                sensor_client_t * client = s->clients;
                while (client) {
                    client->client_callback(client, s->sensor_id, data);
                    client = client->next;
                }
                attr_container_destroy(data);
            }

            s->last_read = now;

            **if (ms_to_next_check == -1 || (ms_to_next_check < s->read_interval))
                ms_to_next_check = s->read_interval;**
        } else {
            int remaining = s->read_interval - elpased_ms;
            **if (ms_to_next_check == -1 || (ms_to_next_check < remaining))
                ms_to_next_check = remaining;**

        }

xwang98, 2021-12-11:
@LinJian1982 good catch! Just wondering if you can create a PR for the fix?

wenyongh, 2021-12-29:
@LinJian1982 Thanks for reporting the issue! The bug was fixed with #917, could you please try again?

closed 2022-02-17 by wenyongh