wasm-micro-runtime/wasm-micro-runtime #893: 'I' cannot represent function return value

LinJian1982, 2021-12-15:
EXPORT_WASM_API_WITH_SIG(wasm_get_microsecond, "()I"),

issue log as below:
[00:06:43:847 - C22A6280]: failed to check signature '()I' and resolve pointer params for import function (env wasm_get_microsecond)

xujuntwt95329, 2021-12-15:
Hi, how did you declare the `wasm_get_microsecond` function in your wasm application?

This log means the function signature mismatched between the wasm module and runtime

LinJian1982, 2021-12-15:
@xujuntwt95329 
uint64 wasm_get_microsecond(wasm_exec_env_t exec_env)

xujuntwt95329, 2021-12-15:
@LinJian1982 Hi, `uint64 wasm_get_microsecond(wasm_exec_env_t exec_env)` is the signature of your native API, I wonder how did you declare the signature of this function in wasm application?

Did you write something like this to declare the function prototype in your application code?
``` c
extern uint64_t wasm_get_microsecond();
```

LinJian1982, 2021-12-15:
@xujuntwt95329
Hi,

I do have following declared in my app header file.  I have the same issue for both host(x86-64) and arm-32 platform.

```
#include "bh_platform.h"

#ifdef __cplusplus
extern "C" {
#endif

/* A time value that is accurate to the nearest
   microsecond but also has a range of years.  */
struct timeval
{
    uint32 tv_sec;		/* Seconds.  */
    uint32 tv_usec;	    /* Microseconds.  */
};

uint64 api_get_microsecond();

```

xujuntwt95329, 2021-12-15:
@LinJian1982 
Hi, which `bh_platform.h` are you using? If you are using `core/shared/utils/bh_platform.h`, then that is for native, and the actual bitwidth of `uint64` is decided by the native OS, it may be incompatible with wasm's type. e.g. the `long int` is treated as 64bit integer on 64bit Ubuntu OS, but it is treated as 32bit integer in wasm.

You can change the definition to
``` c
unsigned long long int 
api_get_microsecond();
```
I think this could fix the problem.

BTW, thanks to your issue, I found that the `uint64_t` definition in our [libc-builtin-sysroot](https://github.com/bytecodealliance/wasm-micro-runtime/tree/main/wamr-sdk/app/libc-builtin-sysroot) also use the wrong bit width, I'll fix it soon and then you can use `uint64_t` in your application file if you use the `libc-builtin-sysroot`.

xujuntwt95329, 2021-12-15:
I've submitted a PR #898 to fix the issue in `libc-builtin-sysroot`

LinJian1982, 2021-12-15:
@xujuntwt95329  
I see there is no definition for uint64 in app sdk, so I defined it as "unsigned long". But looks like the compiler think it is a 32bit platform, so "unsigned long" is 32 bit. 
So I changed the uint64 definition to "unsigned long long", then the issue resolved.

LinJian1982, 2021-12-15:
@xujuntwt95329 

I just see your reply. Accidently, looks like I followed your proposal. 👍 

Jian

closed 2021-12-16 by LinJian1982