wasm-micro-runtime/wasm-micro-runtime #902: iwasm on windows 7 run failed

bxkai, 2021-12-17:
[GetCurrentThreadStackLimits](https://github.com/bytecodealliance/wasm-micro-runtime/blob/5be427bfa26ab4a84e7c1a5dbc8b5e7f9ff5ea86/core/shared/platform/windows/win_thread.c#L569) is not support windows 7

wenyongh, 2021-12-20:
Hi, GetCurrentThreadStackLimits was introduced since Windows 8, could you please test whether the patch works? Thanks.
[get_stack_limits.zip](https://github.com/bytecodealliance/wasm-micro-runtime/files/7745580/get_stack_limits.zip)

bxkai, 2021-12-23:
It's ok， but win7 need rebuild again.I fix it according https://stackoverflow.com/questions/52403473/get-current-used-stack-size-for-win7

wenyongh, 2021-12-29:
@bxkai The patch was merged with #914, could you please try again? Thanks.

bxkai, 2021-12-31:
Implementation of Macro Control，The program compiled by Windows 10 is not compatible with Windows 7. Can the GetCurrentThreadStackLimits function be dynamically loaded? First try get address of GetCurrentThreadStackLimits from "kernel32“. if  not，calling GetCurrentThreadStackLimits_Win7。

wenyongh, 2021-12-31:
Hi, do you wan to build iwasm on Windows 10 and Windows 7? It is not easy to prepare Windows 7 related system to test, could you please help fix it? Thanks.

wenyongh, 2022-01-07:
@bxkai I uploaded patch #939 to fix the issue according to your suggestion, could you please help review and test whether it works? Thanks.

bxkai, 2022-01-25:
I tests，iwasm works well on win7. Thanks

closed 2022-01-25 by bxkai