wasm-micro-runtime/wasm-micro-runtime #926: ./iwasm command Exception

jackiealex, 2021-12-30:
The problem is i use the AOT mode, it do not work!
My steps are below:

1. Create a file test.c with content
`
#include <stdio.h>
#include <stdlib.h>

int main(int argc, char **argv)
{
    char *buf;

    printf("Hello world!\n");

    buf = malloc(1024);
    if (!buf) {
        printf("malloc buf failed\n");
        return -1;
    }

    printf("buf ptr: %p\n", buf);

    sprintf(buf, "%s", "1234\n");
    printf("buf: %s", buf);

    free(buf);
    return 0;
}

// https://giters.com/manis404/wasm-micro-runtime
`
2. compile test.c into test.wasm use command below
emcc test.c -o test.wasm

3. run test.wasm use wasmer:
wasmer test.wasm

its ok. and output is 
Hello world!
buf ptr: 0x500e58
buf: 1234

4. run iwasm with file test.wasm

it not ok, the output is 

[04:54:26:303 - 11C60CE00]: warning: failed to link import function (wasi_snapshot_preview1, fd_close)
[04:54:26:334 - 11C60CE00]: warning: failed to link import function (wasi_snapshot_preview1, fd_write)
[04:54:26:337 - 11C60CE00]: warning: failed to link import function (wasi_snapshot_preview1, fd_seek)
[04:54:26:339 - 11C60CE00]: warning: failed to link import function (wasi_snapshot_preview1, args_sizes_get)
[04:54:26:341 - 11C60CE00]: warning: failed to link import function (wasi_snapshot_preview1, args_get)
[04:54:26:343 - 11C60CE00]: warning: failed to link import function (wasi_snapshot_preview1, proc_exit)
Exception: lookup the entry point symbol (like main, _main, __main_argc_argv) failed

5. try to compile test.wasm to test.aot use wamrc, command is 
wamrc -o test.o test.wasm
and error is

Create AoT compiler with:
  target:        x86_64
  target cpu:    skylake
  cpu features:
  opt level:     3
  size level:    3
  output format: AoT file

Error: invaid llvm binary bin_type.

wenyongh, 2021-12-30:
> The problem is that i use the AOT mode to run a *.aot file, it did NOT work! @wenyongh sir
> 
> My Computer Info(MacOS):
> 
> ```shell
> Darwin MacBook-Pro-4.local 20.6.0 Darwin 
> Kernel Version 20.6.0: Wed Jun 23 00:26:31 PDT 2021; 
> root:xnu-7195.141.2~5/RELEASE_X86_64 x86_64
> 
> CPU
> 2.3 GHz 8-Core Intel Core i9
> ```
> 4. Run `iwasm` command with file test.wasm
> 
> ```shell
> iwasm test.wasm // or ./wasm test.wam
> ```
> 
> But it's NOT ok, the output is
> 
> ```shell
> [04:54:26:303 - 11C60CE00]: warning: failed to link import function (wasi_snapshot_preview1, fd_close)
> [04:54:26:334 - 11C60CE00]: warning: failed to link import function (wasi_snapshot_preview1, fd_write)
> [04:54:26:337 - 11C60CE00]: warning: failed to link import function (wasi_snapshot_preview1, fd_seek)
> [04:54:26:339 - 11C60CE00]: warning: failed to link import function (wasi_snapshot_preview1, args_sizes_get)
> [04:54:26:341 - 11C60CE00]: warning: failed to link import function (wasi_snapshot_preview1, args_get)
> [04:54:26:343 - 11C60CE00]: warning: failed to link import function (wasi_snapshot_preview1, proc_exit)
> Exception: lookup the entry point symbol (like main, _main, __main_argc_argv) failed
> ```
> 

Hi, the LIBC-WASI isn't enabled on darwin platform by default, could you please build iwasm with:
```bash
cmake .. -DWAMR_BUILD_LIBC_WASI=1
```
and rebuild gain?

> 5. Try to compile test.wasm to test.aot using `wamrc`, and my command is
> 
> ```shell
> wamrc -o test.o test.wasm
> ```
> 
> then error is
> 
> ```shell
> Create AoT compiler with:
>   target:        x86_64
>   target cpu:    skylake
>   cpu features:
>   opt level:     3
>   size level:    3
>   output format: AoT file
> 
> Error: invaid llvm binary bin_type.
> ```
> 
> What should I do in order to use AOT mode with **wamrc**

Not sure how you build wamrc, do you build it with `cmake .. -DWAMR_BUILD_PLATFORM=darwin`?
If wamrc doesn't work again, can you please try `wamrc --target-abi=gnu -o test.aot test.wasm`? Thanks.

jackiealex, 2021-12-31:
Yes, i build wamrc with shell below:
```shell
cd wamr-compiler
./build_llvm.sh (or "./build_llvm_xtensa.sh" to support xtensa target)
mkdir build && cd build
cmake .. (or "cmake .. -DWAMR_BUILD_PLATFORM=darwin" for MacOS)
make
# wamrc is generated under current directory
```

jackiealex, 2021-12-31:
And still, wamrc --target-abi=gnu -o test.aot test.wasm cause error:

```shell
Create AoT compiler with:
  target:        x86_64
  target cpu:    skylake
  cpu features:
  opt level:     3
  size level:    3
  output format: AoT file
Error: invaid llvm binary bin_type.
```

event i build wamrc with `cmake ..  -DWAMR_BUILD_PLATFORM=darwin` agagin

jackiealex, 2021-12-31:
use `cmake .. -DWAMR_BUILD_LIBC_WASI=1` well 

iwasm test.wasm runs ok

wenyongh, 2021-12-31:
Hi, normally we use wamrc under Ubuntu/Windows or other Linux systems, I uploaded a patch, could you please patch it, rebuid wamrc and test again?
[wamrc_macos.zip](https://github.com/bytecodealliance/wasm-micro-runtime/files/7795523/wamrc_macos.zip)

jackiealex, 2022-01-04:
git apply wamrc_macos.diff
     
-        if (arch) {
+        if (abi) {
+            /* Construct target triple: <arch>-<vendor>-<sys>-<abi> */
+            const char *vendor_sys;
+            char *arch1 = arch, default_arch[32] = { 0 };
+
+            if (!arch1) {
+                char *default_triple = LLVMGetDefaultTargetTriple();
+
+                if (!default_triple) {
+                    aot_set_last_error("llvm get default target triple failed.");
+                    goto fail;
+                }
+
+                vendor_sys = strstr(default_triple, "-");
+                bh_assert(vendor_sys);
+                memcpy(default_arch, default_triple, vendor_sys - default_triple);
+                arch1 = default_arch;
+
+                LLVMDisposeMessage(default_triple);
+            }
+
+            if (!strcmp(abi, "msvc")) {
+                if (!strcmp(arch1, "i386"))
+                    vendor_sys = "-pc-win32-";
+                else
+                    vendor_sys = "-pc-windows-";
+            }
+            else {
+                vendor_sys = "-pc-linux-";
+            }
+
+            bh_assert(strlen(arch1) + strlen(vendor_sys) + strlen(abi)
+                      < sizeof(triple_buf));
+            memcpy(triple_buf, arch1, strlen(arch1));
+            memcpy(triple_buf + strlen(arch1), vendor_sys, strlen(vendor_sys));
+            memcpy(triple_buf + strlen(arch1) + strlen(vendor_sys), abi,
+                    strlen(abi));
+            triple = triple_buf;
+        }
+        else if (arch) {

it did NOT work either.

wenyongh, 2022-01-04:
@jackiealex did you add `--target-abi=gnu` for wamrc after applying the patch? e.g. `wamrc --target-abi=gnu -o test.aot test.wasm` or `wamrc --target-abi=gnu --disable-simd -o test.aot test.wasm`?

jackiealex, 2022-01-04:
> @jackiealex did you add `--target-abi=gnu` for wamrc after applying the patch? e.g. `wamrc --target-abi=gnu -o test.aot test.wasm` or `wamrc --target-abi=gnu --disable-simd -o test.aot test.wasm`?

Thanks, it works well, i just use aot mode to test performance, and which meets my expectations

closed 2022-02-24 by wenyongh