wasm-micro-runtime/wasm-micro-runtime #928: prinft('%f', 22.00) cause illegal hardware instruction warning

jackiealex, 2021-12-31:
fab.c
```c
int main(int argc, char **argv)
{
  printf("result is %f \n", 22.00); // but if it is   `printf("result is %d \n", 22)` that will ok;
  return 0;
}
```
```shell
emcc fab.c -o fab.wasm
wamrc --cpu=x86-64-v4  --target=x86_64  -o fab.aot  --size-level=1 fab.wasm
iwasm fab.aot
```
output is:
```shell
[1] 85356 illegal hardware instruction  iwasm fab.aot
```

wenyongh, 2021-12-31:
Seems your host cpu isn't `x86-64-v4`, could you please specify the correct cpu name and try again?

jackiealex, 2022-01-04:
```
uname -a //on my host
```
Darwin localhost 20.6.0 Darwin Kernel Version 20.6.0: Wed Jun 23 00:26:31 PDT 2021; root:xnu-7195.141.2~5/RELEASE_X86_64 x86_64

```shell
wamrc --cpu=x86-64  --target=x86_64  -o fab.aot  --size-level=1 fab.wasm
```

Create AoT compiler with:
  target:        x86_64
  target cpu:    x86-64
  cpu features:
  opt level:     3
  size level:    1
  output format: AoT file
Error: SIMD compatibility check failed, try adding --cpu=<cpu> to specify a cpu or adding --disable-simd to disable SIMD

or
```shell
wamrc   -o fab.aot  --size-level=1 fab.wasm
```

Create AoT compiler with:
  target:        x86_64
  target cpu:    skylake
  cpu features:
  opt level:     3
  size level:    1
  output format: AoT file
Error: invaid llvm binary bin_type.

or
```shell
wamrc   -o fab.aot  --size-level=1 fab.wasm
```

Create AoT compiler with:
  target:        x86_64
  target cpu:
  cpu features:
  opt level:     3
  size level:    1
  output format: AoT file
Error: SIMD compatibility check failed, try adding --cpu=<cpu> to specify a cpu or adding --disable-simd to disable SIMD

wenyongh, 2022-01-04:
@jackiealex could you try `wamrc --target-abi=gnu --disable-simd  -o fab.aot fab.wasm` after applying the wamrc patch?

jackiealex, 2022-01-04:
> @jackiealex could you try `wamrc --target-abi=gnu --disable-simd -o fab.aot fab.wasm` after applying the wamrc patch?

I use this, add `--size-level=1`
```shell
wamrc --target-abi=gnu --disable-simd -o add.aot --size-level=1 add.wasm
```
It works well, and thank you

wenyongh, 2022-01-11:
@jackiealex, we made some changes for MacOS platform to make wamrc more easy to use, now normally you can just type `wamrc -o <aot_file> <wasm_file>` to generate AOT file, no need to add other arguments again. Thanks.

closed 2022-02-24 by wenyongh