wasm-micro-runtime/wasm-micro-runtime #932: "Failed to link import function" for pthread functions in C++

VogelIBR, 2022-01-04:
I am trying to use threads in C++. Compiling the C++ code works without any errors. The command for compiling is shown below. The C++ code is a modification of the c pthread example in this repo. The same error occurs when compiling the c pthread example with the `clang++` compiler without any modifications to the code. I am using `wasi-sdk` version `14.0`. 

When trying to run the code I am getting `failed to link import function` errors for the used pthread functions when trying to run code compiled with the C++ compiler. `iwasm` has no problem running the WebAssembly generated by the c compiler.

### Failure with cpp
```
$ /opt/wasi-sdk/bin/clang++ -pthread -O3                \
    -Wl,--shared-memory,--max-memory=196608         \
    -Wl,--allow-undefined,--no-check-features       \
    -Wl,--export=__heap_base,--export=__data_end    \
    thread.cpp -o test.wasm
```
```
$ /home/vogel/code/wasm-micro-runtime/product-mini/platforms/linux/build/iwasm test.wasm
[02:27:23:182 - 7FB3EFFFA740]: warning: failed to link import function (env, _Z12pthread_exitPv)
[02:27:23:192 - 7FB3EFFFA740]: warning: failed to link import function (env, _Z14pthread_createPjPKvPFPvS2_ES2_)
[02:27:23:193 - 7FB3EFFFA740]: warning: failed to link import function (env, _Z12pthread_joinjPPv)
Exception: failed to call unlinked import function (env, _Z14pthread_createPjPKvPFPvS2_ES2_)
```
<details>
    <summary>Code</summary>

#include <pthread.h>
#include <iostream>

void *thread_routine(void *arg)
{
    std::cout << "Enter thread\n";
    pthread_exit(NULL);
    return NULL;
}

int main(int argc, char** argv)
{
    pthread_t tid;
    if (0 != pthread_create(&tid, NULL, thread_routine, NULL)) {
        std::cout << "Failed to create thread\n";
    }
    if (0 != pthread_join(tid, NULL)) {
        std::cout << "Failed to join thread\n";
    }
    std::cout << "Exit\n";
    return 0;
}

</details>

### Working c
```
$ /opt/wasi-sdk/bin/clang -pthread -O3                \
    -Wl,--shared-memory,--max-memory=196608         \
    -Wl,--allow-undefined,--no-check-features       \
    -Wl,--export=__heap_base,--export=__data_end    \
    thread.c -o test.wasm
```
```
$ /home/vogel/code/wasm-micro-runtime/product-mini/platforms/linux/build/iwasm test.wasm
THREAD CREATED 7f92c9038700
Enter thread
Exit
```

<details>
    <summary>Code</summary>

#include <stdio.h>
#include <pthread.h>

void *thread_routine(void *arg)
{
    printf("Enter thread\n");
    pthread_exit(NULL);
    return NULL;
}

int main(int argc, char** argv)
{
    pthread_t tid;
    if (0 != pthread_create(&tid, NULL, thread_routine, NULL)) {
        printf("Failed to create thread\n");
    }
    if (0 != pthread_join(tid, NULL)) {
        printf("Failed to join thread %d.\n", tid);
    }
    printf("Exit\n");
    return 0;
}
</details>

Same faulty behavior when compiling the c version with the `clang++` compiler.

xujuntwt95329, 2022-01-04:
@VogelIBR fixed in #933 
You can pull this PR directly by:
``` bash
git fetch origin pull/933/head:fix932
git checkout fix932
```

or wait for the PR to be merged and pull the main branch.

Thanks a lot!

VogelIBR, 2022-01-04:
Thanks for the quick response!

VogelIBR, 2022-01-05:
I see the pr got merged, closing the issue. Thanks for the fix.

closed 2022-01-05 by VogelIBR