wasm-micro-runtime/wasm-micro-runtime #934: Android platform_init.c and libc_uvwasi.cmake suggestions

xingkaiyu, 2022-01-04:
libc_uvwasi.cmake suggestions:
I see that uvwasi's newest release is v0.0.12 and it use libuv v1.42.0(also the newest version) for default.
I hope you can change the libuv version in libc_uvwasi.cmake to v1.42.0 or get the version from uvwasi.

Android platform_init.c suggestions:
I am building warm with abi armeabi-v7a and ANDROID_PLATFORM android-19, and I found some missing fuction:
ld.lld: error: undefined symbol: accept4
>>> referenced by core.c:498 (/mnt/e/wsl-workspace/WebAssembly/wasm-micro-runtime-WAMR-12-30-2021/wasm-micro-runtime-WAMR-12-30-2021/core/deps/libuv/src/unix/core.c:498)
>>>               core.c.o:(uv__accept) in archive libuv/libuv_a.a

ld.lld: error: undefined symbol: dup3
>>> referenced by core.c:1035 (/mnt/e/wsl-workspace/WebAssembly/wasm-micro-runtime-WAMR-12-30-2021/wasm-micro-runtime-WAMR-12-30-2021/core/deps/libuv/src/unix/core.c:1035)
>>>               core.c.o:(uv__dup2_cloexec) in archive libuv/libuv_a.a

ld.lld: error: undefined symbol: pthread_condattr_setclock
>>> referenced by thread.c:712 (/mnt/e/wsl-workspace/WebAssembly/wasm-micro-runtime-WAMR-12-30-2021/wasm-micro-runtime-WAMR-12-30-2021/core/deps/libuv/src/unix/thread.c:712)
>>>               thread.c.o:(uv_cond_init) in archive libuv/libuv_a.a

ld.lld: error: undefined symbol: epoll_create1
>>> referenced by linux-core.c:88 (/mnt/e/wsl-workspace/WebAssembly/wasm-micro-runtime-WAMR-12-30-2021/wasm-micro-runtime-WAMR-12-30-2021/core/deps/libuv/src/unix/linux-core.c:88)
>>>               linux-core.c.o:(uv__platform_loop_init) in archive libuv/libuv_a.a

ld.lld: error: undefined symbol: epoll_pwait
>>> referenced by linux-core.c:313 (/mnt/e/wsl-workspace/WebAssembly/wasm-micro-runtime-WAMR-12-30-2021/wasm-micro-runtime-WAMR-12-30-2021/core/deps/libuv/src/unix/linux-core.c:313)
>>>               linux-core.c.o:(uv__io_poll) in archive libuv/libuv_a.a

ld.lld: error: undefined symbol: inotify_init1
>>> referenced by linux-inotify.c:74 (/mnt/e/wsl-workspace/WebAssembly/wasm-micro-runtime-WAMR-12-30-2021/wasm-micro-runtime-WAMR-12-30-2021/core/deps/libuv/src/unix/linux-inotify.c:74)
>>>               linux-inotify.c.o:(uv_fs_event_start) in archive libuv/libuv_a.a

I hope you can add them to platform_init.c when __ANDROID_API__ < 21:
int
accept4(int __fd, struct sockaddr *__addr, socklen_t *__addr_length, int __flags)
{
    API_NOT_SUPPORT_ERROR(accept4, 21);
    return -1;
}

int
dup3(int oldfd, int newfd, int cloexec)
{
    API_NOT_SUPPORT_ERROR(dup3, 21);
    return -1;
}

int
pthread_condattr_setclock(pthread_condattr_t *attr, clockid_t clock_id)
{
    API_NOT_SUPPORT_ERROR(pthread_condattr_setclock, 21);
    return -1;
}

int
epoll_create1(int flags)
{
    API_NOT_SUPPORT_ERROR(epoll_create1, 21);
    return -1;
}

int
epoll_pwait(int epfd, struct epoll_event *events, int maxevents, int timeout, const sigset_t *sigmask)
{
    API_NOT_SUPPORT_ERROR(epoll_pwait, 21);
    return -1;
}

int
inotify_init1(int flags)
{
    API_NOT_SUPPORT_ERROR(inotify_init1, 21);
    return -1;
}

wenyongh, 2022-01-06:
@xingkaiyu Thanks for reporting the issue! I uploaded patch #938 to fix it, could you please help review it?

xingkaiyu, 2022-01-10:
I think it is ok now.
And I learned how to fix linux build error when use uvwasi. O(∩_∩)O

closed 2022-01-10 by xingkaiyu