wasm-micro-runtime/wasm-micro-runtime #935: Can call python Module in WAMR

zlonqi, 2022-01-05:
Hello everyone. I implemented some functions in python and need to call this python module in c++ code in WAMR env. But when linking .wasm, the following error occurred:
[image]
the c++ file is:
```c++
//python-call.cpp
#include "/usr/include/python3.6/Python.h"
 #include <malloc.h>
#include <stddef.h>
#include <stdint.h>

#include "Dispatch.h"
#include "KeyValue.h"
#include "Environment.h"
#include "Message.h"
#include "Response.h"
#include "StringArray.h"
#include "Value.h"
#include "WasmExtensions.h"

bool initialize_contract(const Environment& env, Response& rsp)
{
    return rsp.success(true);
}

bool compute_sum(const Message& msg, const Environment& env, Response& rsp)
{
    Py_Initialize(); 
    PyObject* pModule = NULL;
    PyObject* pFunc = NULL;
    PyObject* pName = NULL;
    
    PyRun_SimpleString("import sys");
    PyRun_SimpleString("sys.path.append('/root/private-data-objects/contracts/wawaka/python-call')");
    pModule = PyImport_ImportModule("Handle");//Handle.py contains def HandleFc(creator_id_,block_id,url)
    pFunc = PyObject_GetAttrString(pModule, "HandleFc");//HandleFc can fetch the data from sservice and  compute a result.
    
    PyObject* pArgs = PyTuple_New(3);
    PyTuple_SetItem(pArgs, 0, Py_BuildValue("s", env.creator_id_)); 
    PyTuple_SetItem(pArgs, 1, Py_BuildValue("s", msg.get_string("BlockID"))); 
    PyTuple_SetItem(pArgs, 2, Py_BuildValue("s", msg.get_string("DataURL"))); 

    PyObject* pReturn = PyEval_CallObject(pFunc, pArgs);
    int nResult;
    PyArg_Parse(pReturn, "i", &nResult);
    Py_Finalize();

    ww::value::Number v((double)nResult);
    return rsp.value(v, false);
}

contract_method_reference_t contract_method_dispatch_table[] = {
    CONTRACT_METHOD(compute_sum),
    { NULL, NULL }
};
```
And the main content of  CMakeList.txt is:
```makefile
...
SET(CMAKE_EXECUTABLE_SUFFIX ".wasm")
SET(CMAKE_CXX_COMPILER_TARGET "wasm32")

SET(CONTRACT_INSTALL_DIRECTORY "${PDO_HOME}/contracts")

SET(WASM_BUILD_OPTIONS)
LIST(APPEND WASM_BUILD_OPTIONS "-O3")
LIST(APPEND WASM_BUILD_OPTIONS "-fPIC")
LIST(APPEND WASM_BUILD_OPTIONS "-fno-exceptions")
LIST(APPEND WASM_BUILD_OPTIONS "-nostdlib")
LIST(APPEND WASM_BUILD_OPTIONS "-std=c++11")
LIST(APPEND WASM_BUILD_OPTIONS "-DUSE_WASI_SDK=1")

SET(WASM_LINK_OPTIONS)
LIST(APPEND WASM_LINK_OPTIONS "-Wl,--initial-memory=${LINEAR_MEMORY}")
LIST(APPEND WASM_LINK_OPTIONS "-Wl,--max-memory=${LINEAR_MEMORY}")
LIST(APPEND WASM_LINK_OPTIONS "-z stack-size=${INTERNAL_STACK_SIZE}")
LIST(APPEND WASM_LINK_OPTIONS "-Wl,--allow-undefined")

LIST(APPEND WASM_LINK_OPTIONS "-Wl,--export=ww_dispatch")
LIST(APPEND WASM_LINK_OPTIONS "-Wl,--export=ww_initialize")
...
FUNCTION(BUILD_CONTRACT contract)
  STRING(REPLACE ";" " " WASM_BUILD_OPTIONS "${WASM_BUILD_OPTIONS}")
  STRING(REPLACE ";" " " WASM_LINK_OPTIONS "${WASM_LINK_OPTIONS}")
  ADD_EXECUTABLE( ${contract} ${ARGN} ${WASM_SOURCE} )

  SET(CMAKE_CXX_FLAGS ${WASM_BUILD_OPTIONS} CACHE INTERNAL "")
  SET(CMAKE_CXX_COMPILER_TARGET "wasm32")

  TARGET_INCLUDE_DIRECTORIES(${contract} PUBLIC ${WASM_INCLUDES})
  TARGET_LINK_LIBRARIES(${contract} LINK_PUBLIC ${WASM_LIBRARIES})
ENDFUNCTION()
```
How can I call python functions in c++ code under wamr platform? Any tips will be helpful~Thank you a lot.

xujuntwt95329, 2022-01-06:
Hi, the error message `# error unknown multiarch location for pyconfig.h` is thrown by the preprocessor, it seems that there is no pyconfig file for the target you are building.

Are you cross compiling python into wasm? If so, I think you need to know about how to port python into a new architecture. 

Also, their is another approach, you can build python as native and register some native APIs to achieve the same goals.

zlonqi, 2022-01-06:
Thank you for your advices!
> Are you cross compiling python into wasm? If so, I think you need to know about how to port python into a new architecture.
yes, why is  it a new architecture? My machine is intel x86_64, Ubuntu 18.04TLS. 
> Also, their is another approach, you can build python as native and register some native APIs to achieve the same goals.
Do you mean to directly compile python to wasm?

zlonqi, 2022-01-06:
> Are you cross compiling python into wasm? If so, I think you need to know about how to port python into a new architecture.

I've cancel c++ cross compiling with python into wasm. The python function was reimplemented by C++. But there are other type problems: This seems to be a conflict between wasi-sdk and native linux-sdk, but I haven't figured out how to solve it.Which compilation options need to be set?
```make
make
[  7%] Building CXX object python-call/CMakeFiles/python-call.dir/commonTools.cpp.o
In file included from /root/private-data-objects/contracts/wawaka/python-call/commonTools.cpp:1:
In file included from /root/private-data-objects/contracts/wawaka/python-call/commonTools.h:2:
In file included from /usr/include/x86_64-linux-gnu/curl/curl.h:38:
In file included from /usr/include/x86_64-linux-gnu/curl/system.h:399:
/usr/include/x86_64-linux-gnu/sys/types.h:27:1: error: unknown type name '__BEGIN_DECLS'
__BEGIN_DECLS
^
/usr/include/x86_64-linux-gnu/sys/types.h:86:17: error: typedef redefinition with different types ('__off_t' (aka 'long') vs 'long long')
typedef __off_t off_t;
                ^
/opt/wasi-sdk/share/wasi-sysroot/include/bits/alltypes.h:196:16: note: previous definition is here
typedef _Int64 off_t;
               ^
In file included from /root/private-data-objects/contracts/wawaka/python-call/commonTools.cpp:1:
In file included from /root/private-data-objects/contracts/wawaka/python-call/commonTools.h:2:
In file included from /usr/include/x86_64-linux-gnu/curl/curl.h:38:
In file included from /usr/include/x86_64-linux-gnu/curl/system.h:399:
/usr/include/x86_64-linux-gnu/sys/types.h:109:19: error: typedef redefinition with different types ('__ssize_t' (aka 'int') vs 'long')
typedef __ssize_t ssize_t;
                  ^
/opt/wasi-sdk/share/wasi-sysroot/include/bits/alltypes.h:87:15: note: previous definition is here
typedef _Addr ssize_t;
              ^
In file included from /root/private-data-objects/contracts/wawaka/python-call/commonTools.cpp:1:
In file included from /root/private-data-objects/contracts/wawaka/python-call/commonTools.h:2:
In file included from /usr/include/x86_64-linux-gnu/curl/curl.h:38:
In file included from /usr/include/x86_64-linux-gnu/curl/system.h:399:
In file included from /usr/include/x86_64-linux-gnu/sys/types.h:129:
/usr/include/x86_64-linux-gnu/bits/types/clockid_t.h:7:21: error: typedef redefinition with different types ('__clockid_t' (aka 'int') vs 'const struct __clockid *')
typedef __clockid_t clockid_t;
                    ^
/opt/wasi-sdk/share/wasi-sysroot/include/__typedef_clockid_t.h:4:33: note: previous definition is here
typedef const struct __clockid *clockid_t;
                                ^
In file included from /root/private-data-objects/contracts/wawaka/python-call/commonTools.cpp:1:
In file included from /root/private-data-objects/contracts/wawaka/python-call/commonTools.h:2:
In file included from /usr/include/x86_64-linux-gnu/curl/curl.h:38:
In file included from /usr/include/x86_64-linux-gnu/curl/system.h:399:
In file included from /usr/include/x86_64-linux-gnu/sys/types.h:130:
/usr/include/x86_64-linux-gnu/bits/types/time_t.h:7:18: error: typedef redefinition with different types ('__time_t' (aka 'long') vs 'long long')
typedef __time_t time_t;
                 ^
/opt/wasi-sdk/share/wasi-sysroot/include/__typedef_time_t.h:5:19: note: previous definition is here
typedef long long time_t;
                  ^
In file included from /root/private-data-objects/contracts/wawaka/python-call/commonTools.cpp:1:
In file included from /root/private-data-objects/contracts/wawaka/python-call/commonTools.h:2:
In file included from /usr/include/x86_64-linux-gnu/curl/curl.h:38:
In file included from /usr/include/x86_64-linux-gnu/curl/system.h:399:
/usr/include/x86_64-linux-gnu/sys/types.h:158:20: error: token is not a valid binary operator in a preprocessor subexpression
#if !__GNUC_PREREQ (2, 7)
    ~~~~~~~~~~~~~~ ^
/usr/include/x86_64-linux-gnu/sys/types.h:257:1: error: unknown type name '__END_DECLS'
__END_DECLS
^
In file included from /root/private-data-objects/contracts/wawaka/python-call/commonTools.cpp:1:
In file included from /root/private-data-objects/contracts/wawaka/python-call/commonTools.h:2:
In file included from /usr/include/x86_64-linux-gnu/curl/curl.h:38:
In file included from /usr/include/x86_64-linux-gnu/curl/system.h:405:
/usr/include/x86_64-linux-gnu/sys/socket.h:24:14: error: expected ';' after top level declarator
__BEGIN_DECLS
             ^
             ;
In file included from /root/private-data-objects/contracts/wawaka/python-call/commonTools.cpp:1:
[... 115 lines cut ...]
In file included from /root/private-data-objects/contracts/wawaka/python-call/commonTools.cpp:1:
In file included from /root/private-data-objects/contracts/wawaka/python-call/commonTools.h:2:
In file included from /usr/include/x86_64-linux-gnu/curl/curl.h:38:
In file included from /usr/include/x86_64-linux-gnu/curl/system.h:405:
In file included from /usr/include/x86_64-linux-gnu/sys/socket.h:33:
/usr/include/x86_64-linux-gnu/bits/socket.h:303:35: error: expected function body after function declarator
                                      struct cmsghdr *__cmsg) __THROW;
                                                              ^
In file included from /root/private-data-objects/contracts/wawaka/python-call/commonTools.cpp:1:
In file included from /root/private-data-objects/contracts/wawaka/python-call/commonTools.h:2:
[... cut ...]
         ^~~~~~~~~~~~~~~~~~~~~~
11 errors generated.
python-call/CMakeFiles/python-call.dir/build.make:62: recipe for target 'python-call/CMakeFiles/python-call.dir/commonTools.cpp.o' failed
make[2]: *** [python-call/CMakeFiles/python-call.dir/commonTools.cpp.o] Error 1
CMakeFiles/Makefile2:85: recipe for target 'python-call/CMakeFiles/python-call.dir/all' failed
make[1]: *** [python-call/CMakeFiles/python-call.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2
```

xujuntwt95329, 2022-01-06:
Well, I think you didn't set the toolchain correctly, you are still using host's sysroot.
You should specify the CMAKE_C_COMPILER and CMAKE_CXX_COMPILER to `${WASI-SDK-DIR}/bin/clang` and ${WASI-SDK-DIR}/bin/clang++`

For convenience, you can use this [toolchain file](https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/wamr-sdk/app/wasi_toolchain.cmake) to config these cmake parameters.

BTW, I noticed that you include some **socket** related header files in your code, I'm afraid the network capabilities are not currently supported by WASI-SDK, you may need to do some modification to avoid using socket.

xujuntwt95329, 2022-01-06:
> yes, why is it a new architecture? My machine is intel x86_64, Ubuntu 18.04TLS.

If you are cross compiling python to wasm, then the architecture is wasm32 rather than x86_64

> Do you mean to directly compile python to wasm?

No, I mean implement some native APIs in native to interact with python, and just call these APIs from wasm code.

xujuntwt95329, 2022-01-06:
@zlonqi [Here](https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/doc/export_native_api.md) is a document about how to export native API to wasm application

And [here is a sample](https://github.com/bytecodealliance/wasm-micro-runtime/tree/main/samples/basic) which implement some native APIs and use them in wasm.

**FYI**

zlonqi, 2022-01-06:
> Well, I think you didn't set the toolchain correctly, you are still using host's sysroot. You should specify the CMAKE_C_COMPILER and CMAKE_CXX_COMPILER to `${WASI-SDK-DIR}/bin/clang` and [...]

Thank you xujun. I've tried but failed.

zlonqi, 2022-01-06:
> @zlonqi [Here](https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/doc/export_native_api.md) is a document about how to export native API to wasm application
>
> And [here is a [...]

好的谢谢

zlonqi, 2022-01-06:
Thank you, my friend~ I've solved it. I did repeatedly introduce similar header files.

zlonqi, 2022-01-06:
> BTW, I noticed that you include some **socket** related header files in your code, I'm afraid the network capabilities are not currently supported by WASI-SDK, you may need to do some modification to avoid using socket.

BTW, I link with libcurl.a and wasi-sdk’s libs. But there are new problems: 
[image]
What am I doing wrong here?

xujuntwt95329, 2022-01-06:
I think you are linking a native object file with wasm file?

You can use `file libcurl_la-easy.o` to check the file type, I guess it's `ELF 64-bit LSB relocatable` rather than `WebAssembly (wasm) binary module version 0x1 (MVP)`

zlonqi, 2022-01-06:
> I think you are linking a native object file with wasm file?
> 
> You can use `file libcurl_la-easy.o` to check the file type, I guess it's `ELF 64-bit LSB relocatable` rather than `WebAssembly (wasm) binary module version 0x1 (MVP)`

yes, it's a ELF 64-bit LSB relocatable. Could I trouble you to give some advices? Thank you very much.

xujuntwt95329, 2022-01-06:
ELF 64-bit relocatable and WebAssembly binary module are totally different files, they are not able to be linked together.

For example, if you have `library.c` and `main.c`, then, 
in the **native world**, you can:
``` bash
gcc -c library.c -o library.o  # to get library.o, this is a ELF 64-bit LSB relocatable file
gcc -c main.c -o main.o      # to get main.o, this is a ELF 64-bit LSB relocatable file
gcc library.o main.o -o a.out    # to get a.out, this is a ELF 64-bit LSB shared object
```
The `a.out` is the final executable file, you can `./a.out` to execute it

In the **wasm world**, you can:
``` bash
/opt/wasi-sdk/bin/clang -c library.c -o library.o  # to get library.o, this is a WebAssembly (wasm) binary module
/opt/wasi-sdk/bin/clang -c main.c -o main.o      # to get main.o, this is a WebAssembly (wasm) binary module
/opt/wasi-sdk/bin/clang library.o main.o -o a.wasm   # to get a.wasm, this is a WebAssembly (wasm) binary module
```

The `a.wasm` is the final executable file, you can `iwasm a.wasm` to execute it

However, in your building process, you have a `main.o` which is a `WebAssembly (wasm) binary module`, and a `libcurl_la-easy.o` (from `libcurl.a`) which is a `ELF 64-bit LSB relocatable file`, then the linker (`wasm-ld` in your case) complain that the file format is not valid. This is just like doing this:
``` bash
gcc -c library.c -o library.o  # to get library.o, this is a ELF 64-bit LSB relocatable file
/opt/wasi-sdk/bin/clang -c main.c -o main.o      # to get main.o, this is a WebAssembly (wasm) binary module
wasm-ld library.o main.o -o a.wasm   # error occurs !
```

To solve this problem, you need to have a **wasm version** of `libcurl.a` rather than **native version**, that means you need to have the source file of libcurl, and build it into wasm firstly, and then link it.

But again, I think libcurl will depend on the network capabilities, I'm not sure if it's possible to compile the whole libcurl into wasm without modifying its source code. Also, I'm also curious about your actual usage, if you do need some functionality from libcurl, I think it's better to implement them in native world and expose some high level API to wasm application, rather than build them directly to wasm.

zlonqi, 2022-01-07:
> Also, I'm also curious about your actual usage, if you do need some functionality from libcurl, I think it's better to implement them in native world and expose some high level API to wasm application, rather than build them directly to wasm.

ok. Thank you. the actual usuage is: execute a algorithm which compiled to *.wasm to analyze a dataset in a WAMR of SGX Enclave . And the data to be analyzed needs to be pulled from external data sources.

closed 2022-01-10 by zlonqi