wasm-micro-runtime/wasm-micro-runtime #950: Let's build a python binding

lum1n0us, 2022-01-12:
The WAMR should be used as a library embedded in python. 

References:
- [wasmer-python](https://github.com/wasmerio/wasmer-python)
- [wasmtime-py](https://github.com/bytecodealliance/wasmtime-py)

lum1n0us, 2022-01-12:
Some TODOs:

- Try, learn and investigate the general idea of a Wasm runtime python binding
- Try, learn and investigate [python binding](https://realpython.com/python-bindings-overview/)
- Design the python APIs and list required APIs in C
- Investigate how to structure a Python project, plus how to do test
- Investigate how to publish a python package

BenSunWhiteBoard, 2022-01-18:
Some preliminary result I searched can be find [here](https://github.com/BenSunWhiteBoard/PythonBinding).

lum1n0us, 2022-01-19:
Guess we should focus on *ctype* and *wasmtime-py* in the next phase. That combination is close to our WAMR case. More  questions:
- how to access a C structue in Python ?
- Is it possible to emulte a C pointer in Python ?

Plus, let's try to run some [examples](https://github.com/WebAssembly/wasm-c-api/tree/master/example) with *wasmtime-py*.

lum1n0us, 2022-01-20:
Like, to implement [start.c](https://github.com/WebAssembly/wasm-c-api/blob/master/example/start.c) with *wasmtime-py*

``` python
from wasmtime import Engine, Store, Module, Instance, Trap

print("Initializing...")
engine = Engine()
store = Store(engine)

print("Loading binary...")
with open("start.wasm", "rb") as f:
    binary = f.read()
    print("Compiling module...")
    module = Module(engine, binary)

print("Instantiating module...")
try:
    instance = Instance(store, module, [])
except Trap as trap:
    print("Printing message...")
    print(f">{trap.message}")
    frame = trap.frames
    print("Printing origin...")
```

BenSunWhiteBoard, 2022-01-20:
I did some research and updated the [document](https://github.com/BenSunWhiteBoard/PythonBinding)  for basic usage of ctypes, long story short, it's perfectly fine to use C pointer and C struct in Python with the help of ctypes library. I will carry on and try to run some examples program with wasmtime-py.

lum1n0us, 2022-01-24:
Please try and see if you can access and fork [wamr-python](https://github.com/bytecodealliance/wamr-python).

lum1n0us, 2022-01-24:
@BenSunWhiteBoard there might be an invitation email from Wenyong from github.com in your inbox. Please accept it and let us know any of your concerns.

lum1n0us, 2022-01-24:
It seems there is another way to use *wasmtime-py*. (the code might not be good, but you will see the idea)

```
from ctypes import *
from wasmtime._ffi import *

print("Initializing...")
engine = wasm_engine_new()
store = wasm_store_new(engine)

print("Loading binary...")
with open("start.wasm", "rb") as f:
    wasm = f.read()
    print("Compiling module...")

    binary = wasm_byte_vec_t()
    wasm_byte_vec_new_uninitialized(byref(binary), len(wasm))
    binary.size = len(wasm)
    binary.data = (c_ubyte * len(wasm)).from_buffer_copy(wasm)

module = wasm_module_new(store, byref(binary))
if not module:
    print("> Error compiling module!")

print("Instantiating module...")
imports = wasm_extern_vec_t()
wasm_extern_vec_new_empty(byref(imports))
trap = pointer(wasm_trap_t())
instance = wasm_instance_new(store, module, byref(imports), pointer(trap))
if instance and not trap:
    print("> Error instantiating module, expected trap!")

wasm_module_delete(module)
wasm_store_delete(store)
wasm_engine_delete(engine)

```

The reason I put it here because there are two different forms of APIs
- Procedural Programming. like [c style APIs](https://github.com/WebAssembly/wasm-c-api/blob/master/include/wasm.h)
- Object Orient Programming. like [c++ style APIs](https://github.com/WebAssembly/wasm-c-api/blob/master/include/wasm.hh)

Both can do the job, which is better?

xujuntwt95329, 2022-01-24:
I believe the OOP style one should rely on the C style one, so the C style should be mandatory.

lum1n0us, 2022-01-24:
Totally right.

BenSunWhiteBoard, 2022-01-24:
I can access the repo [wamr-python](https://github.com/bytecodealliance/wamr-python) but I can't fork it. It says "cannot fork because forking is disabled"

lum1n0us, 2022-01-24:
Please try again. It works for me now

BenSunWhiteBoard, 2022-01-24:
Thanks, I can fork it now.

lum1n0us, 2022-01-25:
I've created a [PR](https://github.com/bytecodealliance/wamr-python/pull/3) which is about our design document (docs/design.md).

Please feel free to leave your comments in the PR and take over some interested TODOs especially about *PyPI*.

lum1n0us, 2022-01-25:
> I finished TODOs for passing pointers and PyPI. I will wait for the last PR merged, then I can pull in my fork and create a PR that merge back to main stem, right? I haven't done fork related github operations in a while, just to double check I remeber it right.

you are welcome to push the modifications to my personal repo.

lum1n0us, 2022-01-26:
like
- fork [the branch of my personal repo](https://github.com/lum1n0us/wamr-python/tree/design_doc)
- push local modification and create a PR

BenSunWhiteBoard, 2022-02-15:
I have some thoughts on "TODO: to avoid unnecessary copying between python and native", I think it should be fine if we use ptr and ref to pass the data in and out. The draft below is my understanding towards how ctypes works, I would love to hear your feedback on this.
[image]
And when pass with value, we make a copy in C, no surprise.
[image]

lum1n0us, 2022-02-18:
Have updated the design document PR. please take a look and feel free to comment

lum1n0us, 2022-03-09:
@BenSunWhiteBoard  Hope you are well. Our design document has been reviewed and updated and merged. It is time to start coding. Since it is a test-driven project, all test cases should be finished first. I've finished [cases of `wasm_valtype`](https://github.com/bytecodealliance/wamr-python/pull/5) as an example to express my idea. Please feel free to let me know your ideas on tests if you have any concerns.

If you are going to work on test cases, please call it here first. Just in case we are working on the same one. Please also sync the big function table in design documents with your cases.

I am going to create cases about `wasm_fucntype`.

BenSunWhiteBoard, 2022-03-10:
I will try some unit tests on `wasm_globaltype` first. 

According to my understanding, during this TDD process, we first do unit tests on `xxx_type` because they are more independent and their correctness is kind of foundation of other API? And in later stage we may create more tests for those dependent API(for example `module` or `instance`). I think that's the order for development but I am not sure that's also the order of test too. 

Also, after we are done with unit test, will we create some more complex one like integrated tests or something? I would really appreciate it if you could fill me in with the process we normally would  take when it comes to TDD, just to make sure I understand it correctly, thanks!

BenSunWhiteBoard, 2022-03-11:
I finish `wasm_globaltype`, I will add tests for `wasm_tabletype` next

BenSunWhiteBoard, 2022-03-17:
I finish `wasm_tabletype`, I will add tests for `wasm_memorytype` next, I am glad that I get to know the type system for wasm along the way :)

BenSunWhiteBoard, 2022-03-21:
I finish `wasm_memorytype`, I will add tests for `externtype`

lum1n0us, 2022-04-26:
✋ @BenSunWhiteBoard @xujuntwt95329 @wenyongh @wustwn the python language binding has been almost finished. 

Two PRs,

- https://github.com/bytecodealliance/wamr-python/pull/5 , about unit test cases 
- https://github.com/bytecodealliance/wamr-python/pull/7, about binding generation script

are ready. Please take a look and let me know your opinion.

closed 2022-08-04 by lum1n0us