wasm-micro-runtime/wasm-micro-runtime #961: Inconsistent structure definition invalidating poll_oneoff implementation in libc-wasi

dnitin141, 2022-01-14:
Definition of `__wasi_subscription_t ` is inconsistent with the [wasi-libc](https://github.com/WebAssembly/wasi-libc/blob/659ff414560721b1660a19685110e484a081c3d4/libc-bottom-half/headers/public/wasi/api.h#L1288) definition. This is causing all the blocking calls like `sleep`, `poll`, `ioctl` etc return with an error, instead of blocking. There is no field named [`identifier`](https://github.com/bytecodealliance/wasm-micro-runtime/blob/ee97e30a1ab76eae02435702c9897a5e54ddb052/core/iwasm/libraries/libc-wasi/sandboxed-system-primitives/include/wasmtime_ssp.h#L450) in the [wasi-libc](https://github.com/WebAssembly/wasi-libc) implementation. I could fix it locally by removing the identifier field in the clock struct.

wenyongh, 2022-01-17:
@dnitin141 Thanks for reporting the issue, the wasi-libc definition had changed after
WAMR implemented the libc-wasi support. I tried to fix the issue with PR #964, could
you help test whether it works for you? Thanks.

dnitin141, 2022-01-17:
Yeah, the fix works for me. Thank you!

closed 2022-01-17 by dnitin141